Microchip-Ethernet/EVB-KSZ9477 #2: EVB-KSZ9897 linux git

prabhakarlad, 2017-10-03:
Hi,

We want to access the KSZ9897 using spi interface on X15, We are using the 4.4 kernel. Is there a Linux git repo for KSZ9897, with latest Linux kernel version ?

Is there any documentation on accessing via spi, and merging with latest kernel ? 

Cheers,
--Prabhakar Lad

prabhakarlad, 2017-10-04:
Hi,

I have got the KSZ9897 working with latest kernel 4.14-rc4, now I wanted to test the web page access,
Is there any yocto/package to build this server to install in file system ?

Cheers,
--Prabhakar Lad

prabhakarlad, 2017-10-04:
dmesg
[  637.710166] usb 2-1: new SuperSpeed USB device number 4 using xhci-hcd
[  637.751157] mdio_bus usb-002:004: GPIO lookup for consumer reset
[  637.751243] mdio_bus usb-002:004: using lookup tables for GPIO lookup
[  637.751326] mdio_bus usb-002:004: lookup for GPIO reset failed
[  637.757509] libphy: lan78xx-mdiobus: probed
[  637.879035] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready

wmoors, 2018-10-20:
Bump this back to life...

3.18 LTS has reached end of life in 2017 already. Is there any progress/work on migrating to 4.4?

wmoors, 2018-10-23:
@Ravi-Hegde Excellent news!!

edisonf, 2018-11-30:
Hi @Ravi-Hegde ,

I was wondering how are things going with the release for kernel 4.9 you mentioned by the end of Nov-18? We are planing to use  the KSZ9893, if there is still work to be done maybe we at RidgeRun can also collaborate with the development.