Microchip-Ethernet/EVB-KSZ9477 #4: KSZ8895 Compilation and use

samjenkins00, 2018-04-19:
Hi,

I'm running Linux Yocto 4.1.15 using an NXP i.mx63 ARM processor with the KSZ8895. 
I've managed to get spi-ksz8895 to compile to a .ko, and i can manually modprobe to mount it.

I've read the README but i'm not clear what i need to do to use the switch driver. Do i have to compile and run the sw_setup to setup the switch, and then ksz_sw_sysfs_8895 to create a /spi0.0/sw interface?

ildar1992, 2018-05-10:
Hey,
did someone get how to compile and use this driver with Linux Yocto?

skidaddytn, 2019-04-22:
FWIW, I compiled it into our petalinux 2018.3 distribution (Yocto  w/linux 4.14).   However, while I can see all interfaces with "/sys/bus/spi/devices/spi1.1", I've yet to get it all working.    As for the utilities (regs_bin, kszsw, etc), I had to make them apps and add them to my user layer.  They seem to work, but as I said, I can't get traffic through the switch yet.   

Follow these steps to get where I am:
1)  You will need to make your own yocto kernel layer patch by first modifying the kernel source tree with the files here.   Mod the tree with the cadence and micrel areas, and then run their 2 patches for the ptp and netbridge.  The cadence area is *confusing*...  it would appear that the 4.14 files are not the same-- not sure what to do here, make your own judgement call.  I'm currently experimenting with leaving this area alone, and replacing it with the seemingly older version that Microchip provides.
 After you do this, then make your own patch with "git commit" and then "git format-patch -1".   Stick this patch into your layer (meta-user/recipes-kernel) and reference it in linux-xlnx_%.bbappend.   
2) At this point you can rebuild and should see all the interfaces live in your OS.   To get the utilities, I had to add them as individual user applications (i.e. meta-user/recipes-apps)

If you get this to work, please let me know as I'm doing the same thing... lol

ildar1992, 2019-04-23:
What chip(microprocessor) do you use?

skidaddytn, 2019-04-23:
> 
> 
> What chip(microprocessor) do you use?

Zynq XC7Z020 and we are using the KSZ9567R.   Linux does detect it and work by default without the driver, but to get to the additional features we need the driver working.