Microchip-Ethernet/EVB-KSZ9477 #6: Reset 1PPS state

EmilWine, 2018-07-11:
Hi,

I'm using the ptp4l tool for time syncing between devices. I need to implement a feature where the GPIO_1 output is disabled (i.e. output is 0) and when the ptp4l becomes locked I need the GPIO_1 to output 1PPS in sync with the internal clock. 

By default I get the switch outputting 1PPS on this pin. I tried writing "0" to flag 3 of register 0x052F which disabled the 1PPS output but I could not get it back running by writing "1" to the same flag. Tried toggling flag "2" but with the same result. 

I did notice that restarting my mac driver revives the 1PPS ouptut.
Any suggestions on how to proceed?

Thanks
Emil

wmoors, 2020-03-20:
Bump back to life... 
Did you manage to solve your problem?

triha2work, 2020-03-23:
You cannot change most of the unit (triggered output, timestamp input, and GPIO) related functions directly as the index pointing to those units is likely changed by the driver all the time.  One way to change GPIO is to use the ptpfs sysfs.
echo 0 > .../ptpfs/gpio_1
echo 1 > .../ptpfs/gpio_1