Microchip-Ethernet/EVB-KSZ9477 #12: Booting the EVB-KSZ9477 from SD card

SamirDiwan, 2019-03-05:
Hello,

I am a beginner and new to this community. 

I want to boot the EVB-KSZ9477 board and I followed the instructions given in EVB-KSZ9477 Image Programming Guide for SD card but I am not able to boot the board with it.
After doing SD card Image programming as per the instructions given in the Image Programming guide, when I put SD card in EVB-KSZ9477 SD card slot, powering up the board and connecting it with computer with USB cable and doing the MCU reset button the board is not able to boot with SD card and build root login  terminal doesnot appear.

Am I doing it the right way or am I missing some thing? 

Please guide me with a solution.

Thank you,
Samir

triha2work, 2019-03-05:
Please program the board with a NAND image first if the board is brand-new.  A simple bootstrap firmware is required before U-Boot can be started from the SD card.
Another potential problem is the VARIO Select should be 3V3.  Other selections cause the SD card not working.

SamirDiwan, 2019-03-06:
Thank you for the response. 

Yes the board is brand new and the first initial step that I am implementing is about booting the board with SD card, also tried Flash programming on windows 7 x64  but did not worked. So tried with SD card.

Can you please explain me about the bootstrap firmware and also what are the initial steps for the new board should I implement inorder to make it work and start?

I am sorry if my questions are too basic for you as I am very new to this.

Thank you,
Samir

triha2work, 2019-03-06:
The provided image files are at91bootstrap.bin, u-boot.bin, ubootEnvtFileNandFlash.bin, at91-sama5d3_xplained.dtb, zImage, and rootfs.ubi.  The file at91bootstrap.bin is put first in NAND, then u-boot.bin for U-Boot, ubootEnvtFileNandFlash.bin for U-Boot environmental variables, at91-sama5d3_xplained.dtb for device tree, zImage for kernel, and rootfs.ubi for root filesystem.
In SD card two partitions are created.  A regular DOS one holds boot.bin, u-boot.bin, at91-sama5d3_xplained.dtb, zImage, and uboot.env for the U-Boot environmental variables.  The next linux partition holds the root filesystem.
I do not have problem booting from the SD card by disabling the NAND.  Can you check whether the SD card is programmed properly with the files mentioned?
Also what is your problem of programming the NAND under Windows?  The driver is just not detected and so the script cannot do anything?
And what is the display shown from the board?  Just the "RomBOOT" message?

hardfinder, 2019-09-09:
when  booting from NAND,the ksz9477 board serial puts  error codes. i use 115200 as the rf writes.But why?