Microchip-Ethernet/EVB-KSZ9477 #13: KSZ8863 Support

bnaveen222, 2019-03-05:
Hi,

I am working on Yocto Linux for iMX6 processor. I am looking for Linux drivers for KSZ8863 switch to work with Linux kernel version 4.9. I tried the files available in KSZ/linux-drivers/ksz8863/linux-4.9/drivers/net/ethernet/micrel path. I am facing build errors for this repo. Attached the error log files.

[smiError.txt](https://github.com/Microchip-Ethernet/EVB-KSZ9477/files/2930989/smiError.txt)
[spiError.txt](https://github.com/Microchip-Ethernet/EVB-KSZ9477/files/2930990/spiError.txt)

And I tried to build the kernel available for Atmel board at https://github.com/Microchip-Ethernet/EVB-KSZ9477/tree/master/KSZ/kernels/linux-4.9.143 for KSZ8863, I am getting errors listed in atmelErrorLog.txt

[atmelErrorLog.txt](https://github.com/Microchip-Ethernet/EVB-KSZ9477/files/2931006/atmelErrorLog.txt)

Kindly help in this.

Regards,
Naveen.

triha2work, 2019-03-05:
MRP support should be turned off in kernel configuration as it was implemented in KSZ8863 driver.

bnaveen222, 2019-03-06:
Thank you, it worked.
What about smi mode for KSZ8863, still I'm facing build issues.

triha2work, 2019-03-07:
The sw->mtu code is a mistake.  It should only be used with a certain switch.  You can remove that code for now.

The smi_r and smi_w functions are also wrong.  They should follow the ones implemented in spi-ksz8863.c to call smi_r8 and smi_w8.

Note the KSZ8863 SMI code was never tested as the format used is not standard MDIO format.  The Cadence MDIO controller supposedly can accommodate that, but it was not verified successfully.

bnaveen222, 2019-03-15:
Hi,
I am trying to integrate the Micrel KSZ8863 switch with imx6, imx6 uses fec driver for ethernet, is any fec driver code available for imx6 with ksz8863?

mads-bn, 2019-03-15:
We are considering to use KSZ8563 with iMX8. iMX8 uses fec driver for Ethernet (similar to imx6).
Previously we were able to integrate LAN9354 without touching the fec driver.
I got scared while seeing the amount of conditional KSZ compilation in the provided Atmel reference (macb.c). I don't hope that kind of tight integration is needed? ;-)

We need PTP support.

Previously we used DSA architecture but that looks a bit difficult with this driver/chip?
We will reconsider if we really have to use DSA.
We need link status but we don't have to support STP and we always want L2 bridge in switch.
Later we might need to add VLAN traffic between host and only one of the ports.

bnaveen222, 2019-03-19:
Is porting the fec driver is the only option to work with this chip?
is any other alternatives available?

mads-bn, 2019-03-19:
We used lan9354 based on the standard Linux lan9303 driver, which is a DSA driver (and using MDIO).
We did not need to change fec driver for this mode. We use Linux 4.14.
However I did many changes to make the LAN9354 hw-timestamping work with DSA and a bridged network interface on top of lan1 and lan2. There are many DSA and lan9303 changes after 4.14.
To me it looks like the DSA option for this driver likely not has been maintained for some time.

Currently we investigate if we can use KSZ8563 with iMX8 without using DSA and possibly with a generic PHY driver, since we might not need anything fancy. We will probably only use the PTP delay annotation feature (Correction-field in PTP header) and use hw-timestamping in fec driver (to simplify software task). We have picked I2C because manual mention that MDIO does only provide access to PHY registers and not any switch registers (MDIO worked fine for LAN9354 though and seen same statement for that chip).

bnaveen222, 2019-03-19:
Dear Mads-bn,

I am little confused with your statements and I'm not that much expert related to ethernet switch.

My current scenario is, I'm trying to make ethernet switch working on my board. Initially we connected on SMI interface, later we changed it to SPI because as per above messages the drivers are not tested on SMI.

I have added the driver code available in https://github.com/Microchip-Ethernet/EVB-KSZ9477/tree/master/KSZ/linux-drivers/ksz8863/linux-4.9/drivers/net/ethernet/micrel section to my linux kernel 4.9 path and defined dts files changes to spi mode. Device is detected on spi successfully.

[image]

below is my config file changes:

CONFIG_NET_VENDOR_MICROCHIP=y
CONFIG_KSZ_SWITCHES=y
CONFIG_KSZ_SWITCH=y
# CONFIG_KSZ_DSA is not set
# CONFIG_KSZ_SMI is not set
# CONFIG_KSZ_SWITCH_EMBEDDED is not set
CONFIG_HAVE_KSZ8863=y
CONFIG_HAVE_SPI_KSZ8863=y
# CONFIG_I2C_KSZ8863 is not set
CONFIG_SPI_KSZ8863=y
CONFIG_DEBUG_FS=y 

Without doing any fec changes, my kernel log looks like as below
[image]

Now what should I do to make it work? (You want me to change the fec driver correct?, if yes want the changes required?)

Can we take any conference call to discuss more on this.

Regards,
Naveen.

mads-bn, 2019-03-19:
I don't have any experience yet with KSZ (or with SPI/I2C - only lan9354 with MDIO and DSA).
I think using DSA (CONFIG_KSZ_DSA) is probably not working out of the box.
Will get an eval board soon (KSZ8563) - until then I cannot help with any experience ...

bnaveen222, 2019-03-26:
Hi,

Can you provide the hardware connection between AT91 and KSZ8863(schematics).

Have you tested with iMX6?

Regards,
Naveen.

bvacaliuc, 2019-04-04:
From what I have read of your question we are trying to do the same thing, except with a KSZ9893 device.  We have hardware, but so far our experience is very difficult.  We observed the same as you without modifications to `devices/net/ethernet/freescale/fec.*`.  It did not "just work".

Using the [linux-4.9.143](https://github.com/Microchip-Ethernet/EVB-KSZ9477/tree/master/KSZ/kernels/linux-4.9.143) tree *(same as you)*, we ported it into a [linux-4.9.88 based kernel provided by our SOC vendor](https://github.com/uvdl/linux-imx/tree/imx_4.9.88_2.0.0_ga-var01).

We have tried modifications (both minimal and complete) to the `fec.*`.  We are not resolved yet.  We have gotten as far as getting the interface to accept the switch and I can trigger a scope on TXCTL when I ping the network, but there is no RX activity as far as I can tell.

Based on what I have read, I wonder if it is prudent to require these extensive modifications to the `fec`?  Anyway, we continue going down that path.  I would like also like to explore what can be done using an unmodified `fec` driver.

david-drinn, 2020-09-17:
How did you do over the last year or so, @bvacaliuc? We are interested in using the KSZ8863 ethernet switch with Yocto Linux on the iMX family (possibly iMX 6, 7, or 8). From my reading, it sounds as if that's not going to be easy, but I'm just doing a preliminary look and certainly many of the details here are going over my head.

triha2work, 2020-09-18:
There is unofficial support for KSZ8863 DSA driver if your kernel is new enough.  New DSA drivers are located in drivers/net/dsa/microchip.  They are also located in linux-drivers/dsa/linux-4.19 for a version closer to the version in the current Linux kernel.  For KSZ8863 with 2 ports the performance drop should be acceptable if the host is not doing a lot of TCP transmit.
There are patches for fec_main.c if you want to use the proprietary switch driver.  They are located in linux-drivers/imx.

triha2work, 2020-09-18:
Note the KSZ8563 chip is not in the KSZ8863 family.  It is in the KSZ9893 family, which is similar to KSZ9897 and so can use the same KSZ9897 driver with minor changes.
There is official support for KSZ9897 and KSZ9893 DSA driver with basic switch functions, but 1588 PTP function is not supported yet.

bvacaliuc, 2020-09-18:
Hi Folks,

We did well.  We have successful working designs using the KSZ9893 and KSZ8794 on two different boards.  The kernel we use is currently at this revision: https://github.com/varigit/linux-imx/compare/imx_4.9.88_2.0.0_ga-var01...uvdl:feature/develop

There is the need to run a script at boot to set the RGMII clock delay:
[init-ksz8795](https://github.com/uvdl/debian-var/blob/iris2/patches/init-ksz8795) for the board with the KSZ8794
[init-ksz9897](https://github.com/uvdl/debian-var/blob/iris2/patches/init-ksz9897) for the board with the KSZ9893

We have an unresolved issue that causes the SPI bus to not function unless a gpio-keys is reserved for the kernel.  See https://github.com/uvdl/linux-imx/issues/3  we have not gotten the the bottom of this, but since we have a workaround, we have moved on.

Best Regards,

-bogdan