Microchip-Ethernet/EVB-KSZ9477 #20: KSZ9897 Kernel 4.14 compilation errors

cowen86, 2019-04-15:
I'm compiling the KSZ9897 (a ksz9567 switch) driver for kernel 4.14, but am getting several compilation errors, some of which seem to be reverts from the most recent commit. Was there a reason for these reverts?

Changing from `refcount_set()` back to `atomic_set()`:
https://github.com/Microchip-Ethernet/EVB-KSZ9477/blob/89a73ab19e0f856b82e0971e9a3749a254888c18/KSZ/linux-drivers/ksz9897/linux-4.14/drivers/net/ethernet/micrel/ksz_sw_9897.c#L14373-L14377

Changing back to `*.tv64`:
https://github.com/Microchip-Ethernet/EVB-KSZ9477/blob/89a73ab19e0f856b82e0971e9a3749a254888c18/KSZ/linux-drivers/ksz9897/linux-4.14/drivers/net/ethernet/micrel/ksz_ptp_9897.c#L5584-L5586

Reference to non-existent member `mtu` of the `ksz_sw` struct:
https://github.com/Microchip-Ethernet/EVB-KSZ9477/blob/89a73ab19e0f856b82e0971e9a3749a254888c18/KSZ/linux-drivers/ksz9897/linux-4.14/drivers/net/ethernet/micrel/ksz_sw_9897.c#L8725-L8732

cowen86, 2019-04-15:
Here are the error messages for reference:

| In file included from kernel-source/drivers/net/ethernet/micrel/ksz_sw_9897.c:801:0,
|                  from kernel-source/drivers/net/ethernet/micrel/spi-ksz9897.c:556:
| kernel-source/drivers/net/ethernet/micrel/ksz_ptp_9897.c: In function 'check_sys_time':
| kernel-source/drivers/net/ethernet/micrel/ksz_ptp_9897.c:5584:12: error: request for member 'tv64' in something not a structure or union
|    cur_ktime.tv64 -= ptp->first_ktime.tv64;
|             ^
| kernel-source/drivers/net/ethernet/micrel/ksz_ptp_9897.c:5584:37: error: request for member 'tv64' in something not a structure or union
|    cur_ktime.tv64 -= ptp->first_ktime.tv64;
|                                      ^
| kernel-source/drivers/net/ethernet/micrel/ksz_ptp_9897.c:5585:26: error: request for member 'tv64' in something not a structure or union
|    drift_ktime = cur_ktime.tv64 - passed_nsec;
|                           ^
|   AR      drivers/mtd/mtd.o
|   AR      drivers/net/ethernet/netronome/built-in.o
|   CC      net/ipv6/datagram.o
|   AR      drivers/net/ethernet/qualcomm/emac/built-in.o
|   AR      drivers/net/ethernet/qualcomm/built-in.o
|   AR      drivers/net/ethernet/renesas/built-in.o
|   AR      drivers/net/ethernet/rocker/built-in.o
|   AR      drivers/net/ethernet/samsung/built-in.o
| In file included from kernel-source/drivers/net/ethernet/micrel/spi-ksz9897.c:556:0:
| kernel-source/drivers/net/ethernet/micrel/ksz_sw_9897.c: In function 'sw_init':
| kernel-source/drivers/net/ethernet/micrel/ksz_sw_9897.c:8731:4: error: 'struct ksz_sw' has no member named 'mtu'
|   sw->mtu = sw->reg->r16(sw, REG_SW_MTU__2);
|     ^~
|   AR      net/mac80211/built-in.o
|   CC      drivers/mtd/ubi/cdev.o
| kernel-source/drivers/net/ethernet/micrel/ksz_sw_9897.c: In function 'sysfs_sw_read_hw':
| kernel-source/drivers/net/ethernet/micrel/ksz_sw_9897.c:10335:5: error: 'struct ksz_sw' has no member named 'mtu'
|    sw->mtu = chk;
|      ^~
| kernel-source/drivers/net/ethernet/micrel/ksz_sw_9897.c: In function 'sysfs_sw_write':
| kernel-source/drivers/net/ethernet/micrel/ksz_sw_9897.c:10680:6: error: 'struct ksz_sw' has no member named 'mtu'
|     sw->mtu = num;
|       ^~
|   CC      net/netfilter/core.o
| In file included from kernel-source/include/linux/kernel.h:10:0,
|                  from kernel-source/drivers/net/ethernet/micrel/spi-ksz9897.c:42:
| kernel-source/drivers/net/ethernet/micrel/ksz_sw_9897.c: In function 'sw_check_skb':
| kernel-source/arch/arm/include/asm/atomic.h:31:40: error: 'refcount_t {aka struct refcount_struct}' has no member named 'counter'
|  #define atomic_set(v,i) WRITE_ONCE(((v)->counter), (i))
|                                         ^
| kernel-source/include/linux/compiler.h:264:17: note: in definition of macro 'WRITE_ONCE'
|   union { typeof(x) __val; char __c[1]; } __u = \
|                  ^
| kernel-source/drivers/net/ethernet/micrel/ksz_sw_9897.c:14376:4: note: in expansion of macro 'atomic_set'
|     atomic_set(&sk->sk_wmem_alloc, 1);
|     ^~~~~~~~~~
| kernel-source/arch/arm/include/asm/atomic.h:31:40: error: 'refcount_t {aka struct refcount_struct}' has no member named 'counter'
|  #define atomic_set(v,i) WRITE_ONCE(((v)->counter), (i))
|                                         ^
| kernel-source/include/linux/compiler.h:265:30: note: in definition of macro 'WRITE_ONCE'
|    { .__val = (__force typeof(x)) (val) }; \
|                               ^
| kernel-source/drivers/net/ethernet/micrel/ksz_sw_9897.c:14376:4: note: in expansion of macro 'atomic_set'
|     atomic_set(&sk->sk_wmem_alloc, 1);
|     ^~~~~~~~~~
| kernel-source/arch/arm/include/asm/atomic.h:31:40: error: 'refcount_t {aka struct refcount_struct}' has no member named 'counter'
|  #define atomic_set(v,i) WRITE_ONCE(((v)->counter), (i))
|                                         ^
| kernel-source/include/linux/compiler.h:266:22: note: in definition of macro 'WRITE_ONCE'
|   __write_once_size(&(x), __u.__c, sizeof(x)); \
|                       ^
| kernel-source/drivers/net/ethernet/micrel/ksz_sw_9897.c:14376:4: note: in expansion of macro 'atomic_set'
|     atomic_set(&sk->sk_wmem_alloc, 1);
|     ^~~~~~~~~~
| kernel-source/arch/arm/include/asm/atomic.h:31:40: error: 'refcount_t {aka struct refcount_struct}' has no member named 'counter'
|  #define atomic_set(v,i) WRITE_ONCE(((v)->counter), (i))
|                                         ^
| kernel-source/include/linux/compiler.h:266:42: note: in definition of macro 'WRITE_ONCE'
|   __write_once_size(&(x), __u.__c, sizeof(x)); \
|                                           ^
| kernel-source/drivers/net/ethernet/micrel/ksz_sw_9897.c:14376:4: note: in expansion of macro 'atomic_set'
|     atomic_set(&sk->sk_wmem_alloc, 1);
|     ^~~~~~~~~~
| In file included from kernel-source/drivers/net/ethernet/micrel/spi-ksz9897.c:556:0:
| kernel-source/drivers/net/ethernet/micrel/ksz_sw_9897.c: In function 'sw_stop':
| kernel-source/drivers/net/ethernet/micrel/ksz_sw_9897.c:14681:8: error: 'struct ksz_sw' has no member named 'mtu'
|   if (sw->mtu > 2000) {
|         ^~
| kernel-source/drivers/net/ethernet/micrel/ksz_sw_9897.c:14682:43: error: 'struct ksz_sw' has no member named 'mtu'
|    sw->reg->w16(sw, REG_SW_MTU__2, (u16) sw->mtu);
|                                            ^~
| kernel-source/scripts/Makefile.build:328: recipe for target 'drivers/net/ethernet/micrel/spi-ksz9897.o' failed
| make[6]: *** [drivers/net/ethernet/micrel/spi-ksz9897.o] Error 1
| kernel-source/scripts/Makefile.build:587: recipe for target 'drivers/net/ethernet/micrel' failed
| make[5]: *** [drivers/net/ethernet/micrel] Error 2
| kernel-source/scripts/Makefile.build:587: recipe for target 'drivers/net/ethernet' failed
| make[4]: *** [drivers/net/ethernet] Error 2
| kernel-source/scripts/Makefile.build:587: recipe for target 'drivers/net' failed
| make[3]: *** [drivers/net] Error 2
| make[3]: *** Waiting for unfinished jobs....

embeddman, 2019-10-02:
Hi! Did you successfully compiled drivers on 4.14 kernel? Could you write a shortlist how to port this drivers to new kernel?