Microchip-Ethernet/EVB-KSZ9477 #24: PTP 2-Step Mode?

raveslave, 2019-07-01:
Questions around the KSZ8565R Silicon Errata:  
http://ww1.microchip.com/downloads/en/DeviceDoc/80000751C.pdf

> Module 16: PTP messages get dropped in PTP 2-Step Mode

- Is there a hw-revision of the IC planned to fix this?
- If a software workaround is used, what will be the drawbacks?
- Any interoperability issues due to this?

Ravi-Hegde, 2019-07-01:
At present there is no revisions planned for this chip. The software workaround may cause interop issues. If further assistance needed please contact customer support at https://www.microchip.com/wwwregister/login.aspx?ReturnUrl=https://microchipsupport.force.com

raveslave, 2019-07-01:
Got it, any products using this IC that we can try? 
has it been tested with macOS?