Microchip-Ethernet/EVB-KSZ9477 #30: Linux kernel upgrade

embeddman, 2019-10-02:
Are you planning to upgrade linux kernel to 4.14 version?

triha2work, 2019-10-02:
There is no plan to provide 4.14 kernel directly.  There are 4.14 drivers provided so that they can be used in that kernel.  As the main processor used for this evaluation board is Atmel SAMA5D3, the performance has to be taken into consideration.  It seems somehow the SPI performance is dropped drastically in that version.  I will test whether any version between 4.9 and 4.14 introduces this problem.  As the kernel gets newer it will run into a problem in the Atmel MAC driver, which drastically reduces the network transmit performance.  Of course that MAC driver can be modified to avoid that, but when what is the point of upgrading the kernel except to improve performance or add features?

embeddman, 2019-10-03:
You provide new driver features for 4.14 kernel seperately, but for 4.9 along with the kernel. I'm developing managed switch, and I want to be sure that you will not stop provide new features for 4.9 kernel. As I notice firstly you make changes to 4.14 kernel, than you port them to 4.9. Do you provide all changes for both kernels? As I'm using in my development Atmel SAMA5D36, for me it's bettter to choose 4.9 kernel, isn't it?

triha2work, 2019-11-06:
For now the main kernel is 4.9 and so all fixes and new features are implemented there first.  Only differences in kernels are changes in the kernel APIs.  There are now 4.19 and 5.3 support for the switch drivers.

Letrab, 2019-11-21:
@triha2work : Does 5.3 also suffers prom the SPI performance drop?
If so, any lead on this? Thanks!