Microchip-Ethernet/EVB-KSZ9477 #31: New board, no serial communication

fairchdm, 2019-11-06:
I am new to the KSZ9477 and SAMA5D36.  While reviewing some of the literature it said it would be possible to connect the MPU with a serial cable.  When I connected the cable and booted the KSZ9477 I was only able to see garbage text.  
Do I have to load the Linux software first?  
From the literature I was under the impression the Linux software was pre-loaded.  
Thanks for your help.  
Regards, 
Dennis

triha2work, 2019-11-06:
The baud rate is 115200.

fairchdm, 2019-11-08:
Thanks, that got it.  I had set the baud before, but didn't get comms.  Tried a few other things, switched the RX/TX wires on my serial link but forgot to go back to 115200.