Microchip-Ethernet/EVB-KSZ9477 #34: Unable to setup Switch as 2 step TC

dee101k, 2019-11-21:
Hi all,

I want to run KSZ9477 as 2 step transparent clock. I expect to get correction factor of Followup message to be updated .

However, when i am enabling 2 step mode by setting 515 as 78h then Sync & Follow Up messages are getting dropped and only receiving Announce messages on other ports.

Can anyone explain this type of behavior in KSZ9477 switch ?

wmoors, 2019-11-21:
check errata #20 of the KSZ9477.

dee101k, 2019-12-19:
@wmoors Yes I have gone through #20 of the errata. However, i wanted to setup a switch in IEEE8021.as mode which and also in errata one workaround is mentioned to get same action as 2 step mode by setting switch in 1 step mode.

As mentioned, I have enabled the Egress Sync interrupt to get same timestamp of Sync message in Followup message. However, i am not getting same timestamp. 

Can you guide me as how to do it ?

JosephMatan83, 2021-06-07:
Hi @dee101k,
Did you managed to set the KSZ9477 as a transparent clock ?