Microchip-Ethernet/EVB-KSZ9477 #36: Problems flashing EVB-KSZ9477

danglin44, 2019-12-23:
What I want to do is edit the uboot environment parameters.

First, I can't flash the nand with Ubuntu 18.04.3 LTS:

root@lux0:/home/dave/Documents/Microchip/build/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot# ./flash_board
Executing: sama5d3_xplained O=/home/dave/Documents/Microchip/build/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/images /home/dave/Documents/Microchip/build/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/host/opt/sam-ba/sam-ba /dev/ttyACM0 at91sama5d3x-xplained ./board/atmel/nandflash.tcl -- sama5d3_xplained at91-sama5d3_xplained.dtb video=LVDS-1:800x480-16
-I- Waiting ...
-I- TCL platform : Linux
-I- SAM-BA CDC 2.16  on : linux
no display name and no $DISPLAY environment variable
    while executing
"load /home/dave/Documents/Microchip/build/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/host/opt/sam-ba/sam-ba/lib/tk8.5/../libtk8.5.so Tk"
    ("package ifneeded Tk 8.5.9" script)
    invoked from within
"package require Tk 8.1.1"
    ("package ifneeded BWidget 1.9" script)
    invoked from within
"package require BWidget"
    (file "/home/dave/Documents/Microchip/build/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/host/opt/sam-ba/sam-ba/lib/tkcon.tcl" line 131)
    invoked from within
"source [file join $starkit::topdir lib/tkcon.tcl]"
    (file "/home/dave/Documents/Microchip/build/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/host/opt/sam-ba/sam-ba/main.tcl" line 7)
/buildroot# apt-get install tk8.1.1
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package tk8.1.1
E: Couldn't find any package by glob 'tk8.1.1'
E: Couldn't find any package by regex 'tk8.1.1'

I can flash the image files from windows but so far I haven't discovered the magic to create the file ubootEnvtFileNandFlash.bin.  The file uboot.env is created in a mmc build but not in a nand flash build.  I looked at the two ubootEnvtFileNandFlash.bin files that I have.  However, they don't seem to be standard uboot environment files.

How are they generated?

It may be flash_board generates the default uboot environments...

Letrab, 2020-01-10:
Just copy/paste the contents of the script `./flash_board` in the terminal.

danglin44, 2020-02-19:
The file ubootEnvtFileNandFlash.bin can be generated from a text source file using the command "mkenvimage -s 0x20000 -r".  I manually generated it from the attached text file.

[nand-avb.txt](https://github.com/Microchip-Ethernet/EVB-KSZ9477/files/4226662/nand-avb.txt)