Microchip-Ethernet/EVB-KSZ9477 #39: KSZ8567 and KSZ8873MLL

dayanola, 2020-02-14:
Hello,

I would like to add drivers and device tree entries for KSZ8567 and KSZ8873MLL.
I'm running Linux kernel 4.14.98 on an NXP IMX8 SOM.

Can you please recommend which driver is compatible with these ethernet switches and hopefully a device tree entrée example?

Thanks,
Amit

triha2work, 2020-02-14:
There are examples in at91-sama5d3_xplained_ung8087.dts.

dayanola, 2020-02-14:
Thank you for this information. 

I couldn't find an example for 8567 and 8873. 
Do you know which driver is compatible with them?

triha2work, 2020-02-14:
KSZ8567 is just a variant of KSZ9567, which uses "microchip,ksz9567."  KSZ8873 is about the same as KSZ8863, and uses "microchip,ksz8873."  The only requirements in the device tree are "compatible", "reg", and "spi-max_frequency."  The "ports" is only used for DSA driver.

dayanola, 2020-04-01:
Thanks, the Ethernet switches are connected via I2C. any instructions on that? 
Also , looking at the code here, 9567 is compatible with 9897. (couldn't find 9567 code)

triha2work, 2020-04-01:
KSZ9897 is the main chip with minimum features; KSZ9477 is the chip with maximum features.  Other chips have features in the middle.
I2C is just an access mechanism and there is I2C driver provided, in addition to SPI driver.  The device tree needs to select the correct bus number.  The device created will be #-005f.
In U-Boot I2C support can be enabled to access the chip directly without special command.  Command formats are "i2c md 5f <reg>.2" and "i2c mw 5f <reg>.2 <val>."