Microchip-Ethernet/EVB-KSZ9477 #40: Port remap

embeddman, 2020-02-23:
Hi!
I'm using KSZ9567R, in at91-sama5d3_xplained_ung8071.dts it's ports description like
 `ports {
						#address-cells = <1>;
						#size-cells = <0>;
						port@0 {
							reg = <0>;
							label = "lan1";
						};
						...
					};`
How can I remap port order for example if I want to make 3rd physical switch port like 1st port of device?

triha2work, 2020-02-24:
The port is set by the "reg" parameter.

embeddman, 2020-02-25:
I tried the following config
`ports {
						#address-cells = <1>;
						#size-cells = <0>;
						port@0 {
							reg = <2>;
							label = "lan1";
						};
						port@1 {
							reg = <1>;
							label = "lan2";
						};
						port@2 {
							reg = <0>;
							label = "lan3";
						};
						port@3 {
							reg = <4>;
							label = "lan4";
						};
						port@4 {
							reg = <6>;
							label = "lan5";
						};
						port@5 {
							reg = <3>;
							label = "lan6";							
						};
						port@6 {
							reg = <5>;
							label = "cpu";
							ethernet = <&macb0>;
							fixed-link {
								speed = <1000>;
								full-duplex;
							};							
						};
					};`. As I think reg values <0-6> should correspond to switch ports 1-7. But the upper config doesn't have effect. In sysfs I have the same /sys/class/net/eth0/sw0/0_... for 1-st switch port, not for 3-rd, as I thought. The same effect for dynamic mac table. What does it mean port remap?

triha2work, 2020-02-25:
These settings are only used in DSA driver.
To remap ports you can change port_mappings inside the driver.

closed 2020-02-26 by embeddman