Microchip-Ethernet/EVB-KSZ9477 #45: KSZ9477S with SGMII

hardfinder, 2020-06-02:
Hello,

I plan to use the KSZ9477S switch to connect to the dp83867 phy through the SGMII port. when seteenv  sgmii is 2, after boot,it cannot ping serverip(192.168.0.210),sometimes ping ok just once.

Is it possible? What drivers should I use?

triha2work, 2020-06-02:
sgmii=2 means a copper PHY module is being used, so is this dp83860 phy the PHY module?  It does not seems to be so what is the PHY module being used?  A PHY module looks like this when you search for FCLF8522P2BTL.
If you are integrating a SGMII PHY to the port then it is a direct connection, indicated by sgmii=0.  There is not much support on that configuration as the evaluation board does not have this option.  But customers are able to do that.
One way to debug this problem is to look at the MIB counters.  Using the sysfs mechanism located at /sys/class/net/eth0 or /sys/bus/spi/devices/spi0.0 you can access the switch through files.
Check the port speed: cat sw6/6_speed
In the direct connection case the speed should be fixed at 1000.
Check the MIB counters: cat sw6/6_mib
There are receive and transmit counts.  If there are no transmit counts the switch considers the port is not connected.
If you still have problem using this configuration please open a ticket at Microchip Support to have hardware engineers to look at the hardware configuration.

hardfinder, 2020-06-03:
Thanks for your quickly reply.
The  DP83867 is a phy chip which has a sgmii interface . It is not a module like SFP.
So i want to make port7(sgmii)->>(sgmii)phy(chip)-->>copper(RJ45).

hardfinder, 2020-06-04:
I have finish the issue.