Microchip-Ethernet/EVB-KSZ9477 #53: KSZ8567 2 host ports configuration issue

KarineGeral, 2020-11-04:
Hello 
I'm using ksz9897 linux driver 4.14 on a imx28 based system.
Port 6 and Port 7 are connected to CPU (enet0 and enet1).
I have no modification on FEC driver, only DTS configuration for 2 ethernets using fixed-link-0

Finally, my interface eth0 link is functional, but I can't communicate on eth1 (no ping response).
Does Ksz9897 driver support 2 host ports configuration?
I saw in the source code only host port parameter (sw_host_port)..

Thanks for your support

triha2work, 2020-11-04:
The switch driver is designed to handle one host port, so it will not work with 2 hosts.
The hardware can handle multiple hosts, although it was not fully tested.
It depends on your application.  There are 7 ports in the switch.  Do you want 2 network domains like port 6 is connected to ports 1, 2, 3, and port 7 is connected to ports 4 and 5?  You can use port membership to partition them.  The default value is 0x7f, so you change them with 0x27 for the first domain and 0x58 for the second one.
If you do not need port control then it is the simplest configuration.  You can use the MAC driver as is.
Knowing where the frame is received requires the tail tagging feature.  This can be enabled at each port, so ports 6 and 7 can be enabled at the same time.  This requires modifying the MAC driver to support this feature.
Host port communication problem is typically caused by mismatched RGMII configuration, although in this case both enet0 and enet1 should have the same RGMII settings.  However if the default rgmii_txid needs to be changed to rgmii_id then then software has to change the register properly.  The XMII register is 0xN301 is available at ports 6 and 7 with register 0x6301 and 0x7301.  The default value is 0x08.  It may needs 0x18 to work.
One way to check this problem is to read the MIB counters from both sides.  There should not be any receive errors.  It is a little difficult to read the switch MIB counters though without a switch driver.  The "ethtool -S eth0" command is used to dump the MIB counters from the MAC driver if the function is supported.

KarineGeral, 2020-11-05:
Thank you. In parallel of your response, I've made some tests , modifying the sw_host_port variable, and finally , I'm confused about the usage of this parameter.
By default, when sw_host_port =7, i can't communicate on eth1 network (ENET1 on port7).
When I set sw_host_port =6, I can't communicate on eth0 network (ENET0 on port6), but I can on eth1.
If I set sw_host_port =3 which is a non connected port, both communicates ! eth0 on port 6 and eth1 on port 7 is functionnal.

Can you briefly explains the purpose of this parameter,  and how it works with the hardware physical port connections of the switch?

triha2work, 2020-11-05:
I think the MAC driver is not yet modified to support the switch driver, so network communication does not work when tail tagging is enabled.  You can verify at checking at /sys/bus/spi/devices/spi0.0 (the device name may be different).
cat sw6/6_tail_tag
echo 0 > sw6/6_tail_tag
Turning tail tagging off for the host port should let the MAC communicates properly with the switch again.
KSZ8567 supports PTP, and PTP requires tail tagging, so the driver will enable that feature.  You can manually disable that function with the method above.
The tail tagging feature adds 1-byte tag at the end of the received frame to indicate which port receives it.  For PTP event message another 4-bytes are inserted to indicate the receive timestamp.  When sending the MAC needs to put additional information at the end of the frame for the frame to be successfully sent by the switch.  The basic 2-byte tag contains the port bitmap, transmit queue, and some control bits.  When PTP is enabled another 4-byte transmit timestamp needs to be included all the time.
The frame needs to be at least 60 bytes before adding the tail tag.  For MAC controller that automatically pads the frame up to the minimum Ethernet frame size this will not work.  The padding has to be done by software so that the tail tag can be put at the right place.
With tail tagging off the host port is treated as a regular port and that is why the MAC can communicate with outside just using a fixed-link PHY.
The switch driver tries to use the last port as the host port.  In most cases there is only 1 RGMII/RMII/MII connection and that is the only connection to the MAC.  The newer switches can have more than one such port.  In that case the driver tries its best to find an appropriate host port.  The sw_host_port parameter can be used to override that selection in case it is wrong.
Note the sw_host_port can be set on any port.
As explained before there can be only 1 host port.  If you want to connect 2 you need a special driver.

KarineGeral, 2020-11-06:
I checked, tail tagging is off (sw6/6_tail_tag = 0) although  CONFIG_KSZ_PTP is enabled.

Do you plan a new driver to support 2 host port?

triha2work, 2020-11-06:
I am not sure I follow your response.  Either sw5 or sw6 should have tail tagging on.  If it is on and network communication is not working then turning it off should make it work.
There is no plan to implement a 2 host port driver as there is no valid user case.  You need to propose what you want to do with that setup.  Maybe it is similar to what I mentioned at the beginning.