Microchip-Ethernet/EVB-KSZ9477 #54: KSZ9893 with iMX8m using MDIO

bbyyjjyy, 2020-12-12:
Hi there,

Our team is working on the iMX8mm som from Variscite with the KSZ9893 chip. Based on the suggestion from Variscite support, we modified the board to bypass the AR8033 which on the board by default. Now we need to configure the register on KSZ9893 to remove the delay of RGMII (RGMII_ID_eg).

The KSSZ9893 connected to the board with the MDIO bus and working on the kernel 4.19. We modified the DTS file several times, but no luck, always got the fec module error: eht0 cannot connect to the phy.  
The DTS file basically look like this:
&fec1 {
	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_fec1>;
	phy-mode = "rgmii";
	phy-handle = <&ethphy0>;
	fsl,magic-packet;
	status = "okay";
	phy-supply = <&reg_ethphy0>;
	phy-reset-gpios = <&gpio1 2 GPIO_ACTIVE_LOW>;
	mdio {
		#address-cells = <1>;
		#size-cells = <0>;

		ethphy0: ethernet-phy@1 {
			compatible = "ethernet-phy-ieee802.3-c22";
			reg = <1>;
		};
	};
};

Is it possible to configure the register through mdio bus? Is there any driver or patch we need to install to the kernel?

Did anyone get any ideas?

Please help.

Thanks
Jacob

triha2work, 2020-12-14:
There are 2 KSZ9893 drivers: non-DSA and DSA.  DSA driver is maintained in the mainline kernel and does not require modification in the host MAC driver, but it only has basic functionality.  Non-DSA driver requires significant changes in the MAC driver depending on the configuration.
The switch port that connects to the MAC is called the host port.  The connection is fixed and so there is no actual PHY device.  The device tree just specifies fixed-link with a speed so that a fixed PHY device will be used.
The non-DSA driver simulates the switch as a PHY so that the MAC driver can connect it through a virtual PHY device.  The MAC driver can automatically get the connected speed if the interface is RMII or MII, but for RGMII there are delay settings that are out of the control of the switch driver.  Those settings have to be set on the MAC side and so in the device tree.  The default setting is rgmii_txid.  The right setting probably needed is rgmii_id.
However, it seems the error happens because the switch driver is not loaded.  You need to check that first.  The SPI version is normally loaded before the MAC driver,  but the I2C version is probably after it.
It is probably not right to use the term MDIO for the bus connected between the MAC and the virtual PHY.  There are no PHY registers to access as it is a pure hardware interface.  Software has no control to make it work except setting the right RGMII delays.
If the switch registers can be accessed either in U-Boot or through the switch driver then the register 0x3301 has to be modified to make this interface works.  The default value is 0x4b.  It probably needs to be changed to 0x5b.
If the Freescale driver fec_main is used then there is a patch to make it work with the switch driver.

triha2work, 2020-12-15:
The correct term to use is media-independent interface (MII).  MDIO defines a format for the MAC to access PHY registers.  Some switches provides SMI, a mechanism to access switch registers through MDIO bus, albeit sometimes in a non-standard format.
For example, one switch uses an external PHY which registers can be accessed through MDIO bus, but the PHY's RGMII is only connected inside the switch port.

JacobBai, 2020-12-16:
Thanks for the reply!

I will try it out and let you know how it goes.

Jacob

kmeaw, 2022-09-17:
Hello, @JacobBai. Have you figured out what the problem was?

bbyyjjyy, 2022-09-18:
> Hello, @JacobBai. Have you figured out what the problem was?

We ended up using NXP iMX8mm SOM with KSZ9893. But I believe the problem was about a capacitor on the Variscite board added extra delay onto RGMII line (tx or rx, don’t remember). Try different RGMII delay on the register with capacitor removed and not removed board. Good luck!