Microchip-Ethernet/EVB-KSZ9477 #55: KSZ9477 has no extenal phy support ?

nagesh-sonwane, 2020-12-24:
I am using KSZ9477 chip connected with NXP iMX6xSolo2 Processor. The hardware design contains external phy to be connected to internal phy of siwtch. Looking at the code, there seems internal phy driver and I am not able to attach external phy with driver code and not with dts also.
My question is how  I can make the support for external phy ? what **probable modification** are required in switch driver code and how much  time we should consider/expect in general for that ?

triha2work, 2021-01-04:
The switch does not have direct access to the external PHY so the driver has to assume the link is up.  The external PHY registers may be accessible through MDIO from the MAC driver, but there is no good way for the I2C/SPI switch driver to do that.

nagesh-sonwane, 2021-01-06:
Thanks for the reply. Though 'Switch' do not have direct access to external phy; the (MAC) driver does as you mentioned. The current (ksz9477) driver implementation has modifications to MAC driver (macb for Atmel .. fec for i.MX e.g.). This way, instead of attaching switch ports interfaces (net_devs) to internal mdio bus (switch virtual MDIO bus), we can attach external phy to (fec) MAC mdio bus and leverage that bus for switch driver for external phy support.
I am thinking if this is possible and can we do this or do you feel any issue in such implementation.

The another question I have is I think DSA supports external phy support, cant we implement like that ?

Note that I am using 4.14 driver version which doesnt seems to be leveraging DSA/Switchdev architecture (where 4.9 kernel seems). Please correct me here if this is not right understanding as such.

triha2work, 2021-01-07:
I do not know much about DSA and device tree to verify it can be done.  It is probable the PHY can be specified for the port so DSA gets the PHY register information through the PHY device instead of asking through the switch driver.
For the non-DSA driver it is also probable but it requires special code.  Right now the switch driver tells the modified MAC driver how many network devices it needs to create.  After that each network device connects to a virtual PHY device created by the switch driver.  For the port using external PHY then the MAC driver needs to connect to a PHY device created on the real MDIO bus.
Note that the main PHY device driver will take control of that PHY and it needs special attention of the adjust_link helper function to ignore any information sent from that port.