Microchip-Ethernet/EVB-KSZ9477 #57: KSZ8567STXI MDIO Linux Bring Up

topsecretmed, 2021-02-19:
Hello, 

I was wondering if you could point me in the correct direction for setting up the KSZ8567STXI phy or similar product phy? I searched the entire repo for this particular phy example with no particular luck. 

I'm looking to interface the phy using MDIO and I have ports 1 and 3 connected to a device.

triha2work, 2021-02-19:
KSZ8567 is in the KSZ9477/KSZ9897 family of switches.  It is a variant of KSZ9567 and has 100Mbit ports instead of gigabit.
The switch registers can be accessed through SPI or I2C.  I am not sure the switch can be accessed through MDIO.
The host port of the switch is connected to the MAC.  The MAC driver can use the fixed PHY to connect to the switch for basic network communication.
However, it seems the STXI designation means the host port is using SGMII connection.  That will require the switch driver to setup some registers for that connection to work if the MAC is directly connected.

topsecretmed, 2021-02-19:
Hey,

Thanks for the reply! I'm a bit lost in your explanation, but I will highlight what I understand so you can correct me :) 

To give more context of how we're using the chip, we wish to send packets out of port 1 and 3, but port 1 and 3 won't be communicating with each other. With that being said, from what I've read in the KSZ8567 datasheet, we should be able to send the packets out the correct port based on the learned MAC address table from the single SGMII connection? I'm used to working with PHYs that have separate SGMII interfaces for each ethernet port on a chip. 

As an example, I'd imagine eth0 coming up in Linux, and after some ARPs, the chip determines the MAC addresses associated with each port. If I send a UDP packet out to eth0, the KSZ8567 will know which port to send the packet out? 

If that is the case, does that mean the default switch registers on start up is okay for our use case? We're not planning on doing anything fancy (vlans, QoS, etc...) other than sending packets in and out of port 1 and 3. We're only looking at the MDIO interface for link, speed, and duplex status.

triha2work, 2021-02-19:
It depends on what IP subnets are used in devices connected to port 1 and 3.  Typical implementation is to create network device eth0 for port 1 and eth1 for port 3.  If different subnet is used then the network stack knows which network device to use and everything is controlled by software.  If the same subnet is used then the routing table dictates which network device to send the response so there is no control which port is used inside the stack.  The user applications still have control over which port to use.
This implementation requires the tail tagging feature which requires modification of the MAC driver to interact with the switch.
A simple implementation of not using tail tagging is to manipulate the port membership so that port 1 and 3 does not talk to each other.  Frame receiving from these ports can just be passed straight to the host port.  Unicast frames sent from the host port will go to the correct port if the address is learned.  Multicast and broadcast frames and first unicast frame that is not learned will come out both ports.
For the first implementation you can look at the DSA driver that is available in the mainline kernel depending on which version you want to use.