Microchip-Ethernet/EVB-KSZ9477 #66: error: #error "Please port gnulib freadahead.c to your platform!

aep, 2021-04-12:
EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot  does not compile for me

is there a binary available?

```

/usr/lib/ccache/bin/gcc  -I.   -I/home/aep/kram/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/host/usr/include  -O2 -I/home/aep/kram/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/host/usr/include -c -o glthread/lock.o glthread/lock.c
freadahead.c: In function 'freadahead':
freadahead.c:91:3: error: #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."
   91 |  #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."
      |   ^~~~~
make[4]: *** [Makefile:1842: freadahead.o] Error 1
make[4]: *** Waiting for unfinished jobs....
fseeko.c: In function 'rpl_fseeko':
fseeko.c:109:4: error: #error "Please port gnulib fseeko.c to your platform! Look at the code in fseeko.c, then report this to bug-gnulib."
  109 |   #error "Please port gnulib fseeko.c to your platform! Look at the code in fseeko.c, then report this to bug-gnulib."
      |    ^~~~~

```

wmoors, 2021-04-15:
Bump! 

I am having the same issue.

ljouanet, 2021-06-09:
I have the same result despite update m4 package like suggested in
https://askubuntu.com/questions/1099392/compilation-of-m4-1-4-10-to-1-4-18-fails-due-to-please-port-gnulib-freadahead-c
Also, you can find demo image EVB-KSZ9477/9563 on https://github.com/Microchip-Ethernet/EVB-KSZ9477/releases/tag/v1.2.1

triha2work, 2021-09-22:
The updated m4 package still does not resolve the problem in your environment?  What is the Linux distribution you are using?  Note the updated buildroot can build the image under Ubuntu 20.04 in WSL.