Microchip-Ethernet/EVB-KSZ9477 #67: Not able to configure time aware shaper

joschi2804, 2021-04-13:
Hello,

does KSZ9477 not support time aware shaper?
I tried to set tas qdisc with tc command:
`tc -d qdisc replace dev eth0 parent root handle 100 taprio num_tc 2 map 1 1 1 1 1 0 1 1 1 1 1 1 1 1 1 1 queues 1@0 1@1 base-time 1618273428 sched-entry S 01 300000 sched-entry S 02 300000 clockid CLOCK_TAI`

I got "_Unknown qdisc "taprio", hence option "num_tc" is unparsable_" error. 
The kernel version is 4.9.143

If it's not working with tc tool, is it maybe possible to enable time aware shaper with the internal switch registers? If yes, how can i enable them?

BR

triha2work, 2021-04-27:
The driver does not support the tc command.  The sysfs mechanism is used to configure the switch.  Look for the SPI device at /sys/bus/spi/devices/spi1.0.  The device name comes from the bus and chip select.  There are sw and several sw# with a number, which are used for port configuration.  Use "echo 1 > sw/info" to turn on verbose mode.  Use "cat <file>" to get some information from the file.
Use the "avb=1" driver parameter to enable most of the AVB functions in the driver.  After that go into the appropriate port directory like sw1.
Check the tc map with "cat 1_tc_map" which shows which VLAN priority maps to which queue.
"echo 3 > 1_q_index" to select queue 3.  Check the current setting with "cat 1_q_scheduling" and "cat 1_q_shaping."  The bandwidth then can be set with "echo 10000 > 1_q_admin_mbps" for 10 Mbps.  Send traffic to port 1 to come out of port 2.  Check the MIB counters of port 2 with "cat 1_mib."  There should be tx_discards number.

kcn21, 2021-10-18:
> Hello,
>
> does KSZ9477 not support time aware shaper? I tried to set tas qdisc with tc command: `tc -d qdisc replace dev eth0 parent root handle 100 taprio num_tc 2 map 1 1 1 1 1 0 1 1 1 1 1 1 1 1 1 1 [...]

Yes @joschi2804 , Same error for me. Do you know the solution ? @triha2work  there is no driver problem at my end it supports tc commands.