Microchip-Ethernet/EVB-KSZ9477 #74: KSZ9477 : multi_dev=1 on FEC

jeghub, 2021-07-21:
Hi,

I need your help, I'm still trying to get the driver work on FEC (imx7d) with multi-dev=1.
I'm using KSZ9477 chip,  with multi_dev set to 1 on kernel 4.14. 
Here are my boot settings : 
`spi-ksz9897.multi_dev=1 spi-ksz9897.iba=0 spi-ksz9897.sgmii=1 spi-ksz9897.stp=0 spi-ksz9897.avb=0`
Work fine on our custom board with multi-dev = 0 with the same settings.

I'm using the fec_main.c and fec patch from this git : [imx-linux-4.14](https://github.com/Microchip-Ethernet/EVB-KSZ9477/tree/master/KSZ/linux-drivers/imx/imx-linux-4.14/drivers/net/ethernet/freescale) 
Try with files from linux-4.14, same issues.

- First problem appeared when using iperf3 tool. After some seconds, some ports does not work anymore and Tx ring dump is write indefinitely in serial console. This occured in normal use but it take a lot more time. With iperf I got it in 3 seconds. 

- Second problem, I got kernel oops du to bad memory free on skb (see attached file [kOops_multidev1.txt](https://github.com/Microchip-Ethernet/EVB-KSZ9477/files/6853714/kOops_multidev1.txt)) after some random time (minutes to hours)

I've made a change that seems to correct the first issue with iperf : in `fec_enet_rx_queue(struct net_device *ndev, int budget, u16 queue_id)`, the ndev pointer is no longer use directly, a local var is use instead so ndev is not changed.
Find attached a patch for im-linux-4.14 fec_main.c : [fec_main_tx_ring_dump.patch.txt](https://github.com/Microchip-Ethernet/EVB-KSZ9477/files/6853967/fec_main_tx_ring_dump.patch.txt)

parthshah3690, 2021-09-17:
Bump!

I am having the similar issue. Please check #77 .
Do you have a solution already?

jeghub, 2021-09-27:
Please find attached :
[dsti.txt](https://github.com/Microchip-Ethernet/EVB-KSZ9477/files/7234590/dsti.txt)

parthshah3690, 2021-10-12:
Hi @jeghub ,
Are you able to solve the crash with attached dtsi file?

jeghub, 2021-10-12:
No. For me the device tree is not related to this bug. 
For the iperf bug, see my first comment. I had error that lead to Tx_rind_dump and the crash.
```
I've made a change that seems to correct the first issue with iperf : in fec_enet_rx_queue(struct net_device *ndev, int budget, u16 queue_id), the ndev pointer is no longer use directly, a local var is use instead so ndev is not changed.
Find attached a patch for im-linux-4.14 fec_main.c : fec_main_tx_ring_dump.patch.txt

```
With this patch, I'm able to use iperf with the chip in multi-dev=0;

My problem is that for some application I need to have an iface for each port (that's what the driver do when set in multi-dev=1) but in this mode, I've kernel OOPS at random times. That's this issue with multi-dev=1 that I'm not able to resolve. (And I've the same issue (bad memory free that leads to kernel oops) when enabling STP in multi-dev=0 mode.

Bartel-C8, 2022-03-29:
@jeghub , Sorry for highjacking your issue. But do you have maybe some KSZ chips on hand that you want to sell? We are looking for a few... Thanks in advance!