Microchip-Ethernet/EVB-KSZ9477 #75: Avoid patching kernel to make KSZ8895 SPI driver compile

emilm, 2021-07-21:
I see that the kernel maintainers have removed 

```
int skb_append_datato_frags(struct sock *sk, struct sk_buff *skb,
			    int getfrag(void *from, char *to, int offset,
					int len, int odd, struct sk_buff *skb),
			    void *from, int length);

```

https://github.com/Microchip-Ethernet/EVB-KSZ9477/blob/master/KSZ/linux-drivers/ksz8895/linux-5.3/include/linux/skbuff.h#L1151

Is it possible to change the code to work without this function? Thanks. I am on kernel 5.4

triha2work, 2021-09-22:
There are updated drivers for 5.4 and 5.10 in linux-drivers.

emilm, 2021-09-22:
Thanks @triha2work was it added 12 days ago?

triha2work, 2021-09-28:
The switch drivers were updated, and then the MAC driver patches were updated to use the mainline kernels instead of the ones from linux4sam.

emilm, 2021-09-29:
It does not compile for 5.4:

ksz_sw_8895.c:3686:11: error: 'struct ksz_port_info' has no member named 'lpa'

emilm, 2022-03-24:
Any idea?