Microchip-Ethernet/EVB-KSZ9477 #82: KSZ8895 not working with Linux v5.10 from NXP's IMX BSP

pedwo, 2021-12-23:
When integrating the Microchip v5.10 switch driver with NXP's BSP v5.10 kernel [lf-5.10.72-2.2.0](https://source.codeaurora.org/external/imx/linux-imx/tag/?h=lf-5.10.72-2.2.0), the switch doesn't work. It probes ok, the interfaces come up, but ping doesn't work.
Note that everything is ok with the Microchip v5.4 driver on NXP's BSP v5.4 kernel [rel_imx_5.4.70_2.3.0](https://source.codeaurora.org/external/imx/linux-imx/tag/?h=rel_imx_5.4.70_2.3.0).
Since the NXP BSP is used by a lot of people, it would be great if you can check this.

Bartel-C8, 2022-08-11:
Same problem as I found out here?

https://github.com/Microchip-Ethernet/EVB-KSZ9477/issues/63#issuecomment-971431494

Try to remove spi-cpha from the DTS for the KSZ

triha2work, 2022-08-11:
Probably not as the SPI access is stable throughout the kernels.  A major difference between 5.4 and 5.10 is the phylink API was changed again.  The phylink code was verified on the KSZ9897 driver and then ported to other switch drivers.  The exact KSZ8895 driver was not verified with the 5.10 kernel, so there may be some incompatibilities.  For network communication issues please check the MIB counters of the MAC and the switch and make sure the link is reported correctly.

mkphsnbng, 2023-03-09:
@triha2work @Bartel-C8 

I am facing similar issue,
i took 5.10 KSZ8863(https://github.com/Microchip-Ethernet/EVB-KSZ9477/tree/master/KSZ/linux-drivers/ksz8863/linux-5.10/drivers/net/ethernet/micrel)  and used in 5.10 kernel which comes with SDK 8.2.0.24 and observed that ping is not happening and also 
i am find that few logs related to phy is not showing while booting..
I raised a query in the same gilt blog below is the link please have a look, Also may i know what are the changes it may require in dts/driver/defconfig?

https://github.com/Microchip-Ethernet/EVB-KSZ9477/issues/91