Microchip-Ethernet/EVB-KSZ9477 #89: Use KSZ9896 as PHY on Jetson AGX Xavier

lantianhaohh, 2023-01-16:
Use the KSZ9896 PHY to replace the Jetson-AGX-Xavier PHY driver via the MDIO interface,I modified the device tree to fixed-link mode, modified as follows:

```
ethernet@2490000 {
		compatible = "nvidia,nveqos";
		reg = <0x00 0x2490000 0x00 0x10000>;
		reg-names = "mac-base";
		interrupts = <0x00 0xc2 0x04 0x00 0xba 0x04 0x00 0xbe 0x04>;
		clocks = <0x04 0x120 0x04 0x20 0x04 0x22 0x04 0x21 0x04 0x23 0x04 0x08>;
		clock-names = "pllrefe_vcoout\0eqos_axi\0eqos_rx\0eqos_ptp_ref\0eqos_tx\0axi_cbb";
		resets = <0x04 0x11>;
		interconnects = <0x03 0x8e 0x03 0x8f>;
		interconnect-names = "dma-mem\0dma-mem";
		reset-names = "mac_rst";
		nvidia,promisc_mode = <0x01>;
		nvidia,num-dma-chans = <0x01>;
		nvidia,dma-chans = <0x00>;
		nvidia,slot_num_check = <0x00 0x00 0x00 0x00>;
		nvidia,slot_intvl_vals = <0x00 0x7d 0x7d 0x7d>;
		nvidia,num-mtl-queues = <0x01>;
		nvidia,mtl-queues = <0x00>;
		nvidia,rx-queue-prio = <0x02>;
		nvidia,dcs-enable = <0x01>;
		nvidia,pad_calibration = <0x01>;
		nvidia,tx-queue-prio = <0x00>;
		iommus = <0x02 0x14>;
		iommu-resv-regions = <0x00 0x00 0x00 0x40000000 0x00 0x60000000 0xffffffff 0xffffffff>;
		dma-coherent;
		nvidia,rx_riwt = <0x100>;
		nvidia,rx_frames = <0x40>;
		nvidia,tx_usecs = <0x100>;
		nvidia,tx_frames = <0x05>;
		nvidia,ptp_ref_clock_speed = <0x12a05f20>;
		status = "okay";
		pinctrl-names = "mii_rx_disable\0mii_rx_enable";
		pinctrl-0 = <0x1f>;
		pinctrl-1 = <0x20>;
		nvidia,rxq_enable_ctrl = <0x02>;
		nvidia,pause_frames = <0x00>;
		nvidia,phy-reset-gpio = <0x0b 0x35 0x01>;
		phy-mode = "rgmii-id";
		nvidia,max-platform-mtu = <0x3fff>;
		vddio_sys_enet_bias-supply = <0x21>;
		vddio_enet-supply = <0x21>;
		phy_vdd_1v8-supply = <0x22>;
		phy_ovdd_rgmii-supply = <0x22>;
		phy_pllvdd-supply = <0x21>;

		eqos-cool-dev {
			cooling-min-state = <0x00>;
			cooling-max-state = <0x05>;
			#cooling-cells = <0x02>;
			phandle = <0x28f>;
		};

		prod-settings {
			#prod-cells = <0x04>;

			prod {
				prod = <0x00 0x8800 0x80000000 0x00 0x00 0x8804 0x20000000 0x20000000>;
			};
		};
               //add
		fixed-link {
			speed = <0x3e8>;
			full-duplex;
		};

		mdio {
			compatible = "nvidia,eqos-mdio";
			#address-cells = <0x01>;
			#size-cells = <0x00>;
                        //add
			// phy@0 {
			// 	compatible = "microchip,ksz9477";
			// 	reg = <0x00>;
			// 	interrupt-parent = <0x0b>;
			// 	interrupts = <0x34 0x08>;
			// 	microchip,copper-mode;
			// 	microchip,reg-init = <0x03 0x12 0x7fff 0x880>;
			// 	phandle = <0x2e7>;
			// };
			//add
				ksz9477: ksz9477@0 {
					compatible = "microchip,ksz9477";
					reg = <0x00>;
					interrupt-parent = <0x0b>;
					interrupts = <0x34 0x08>;
					status = "okay";
					ports {
						#address-cells = <1>;
						#size-cells = <0>;
						port@0 {
							reg = <0>;
							label = "lan1";
						};
						port@1 {
							reg = <1>;
							label = "lan2";
						};
						port@2 {
							reg = <2>;
							label = "lan3";
						};
						port@3 {
							reg = <3>;
							label = "lan4";
						};
						port@4 {
							reg = <4>;
							label = "lan5";
						};
						port@5 {
							reg = <5>;
							label = "cpu";
							ethernet = <&macb0>;
							phy-mode = "rgmii-txid";
							fixed-link {
								speed = <1000>;
								full-duplex;
								pause;
							};
						};
						port@6 {
							reg = <6>;
							label = "lan6";
						};
					};
				};
		};
	};
```
However, the network has not started successfully, is the configuration of fixed-link normal, and do I need to modify other driver codes? Can you help? How to make KSZ9896 work on jetson.My board is connected in MDIO, not SPI and I2C.

sgidel, 2023-01-21:
Which driver are you trying to use? The DSA driver or the custom microchip driver that requires adding support for the specific mac you are using?
If you want to avoid having to write a bunch of code yourself, use the in-tree upstream DSA driver.
As far as I can tell neither of these drivers has support for the MDIO configuration mode and you will need to connect the SPI or I2C interface.
I implemented the KS9477 on an NVIDIA Orin platform and ran into a few problems:
- The `tag_ksz` driver causes a kernel panic due to occasionally receiving skbs that have no tailroom.
I solved this by adding skb_pad to the appropriate xmit function. Its a hack but it works and doesnt cause any performance issues. 
- There is a bug in the `dsa_core` module for the kernel versions that nvidia uses that will cause significant packet loss on tcp transmit. See my response on this other issue - #86 
- Nvidia's new nvethernet driver in L4T 35+ has a bug that sets the MTU lower than specified because it always thinks macsec is enabled and wont let you change the MTU if the interface is up. This doesn't seem to cause problems but does result in some kernel log spam at startup.

Your device tree as-is is mostly valid. You will need to move the entire ksz9477 portion to under the i2c/spi node you end up using. You also need to make sure the ethernet parameter in the port@5 node points to a label assigned to the ethernet controller.

lantianhaohh, 2023-01-29:
@sgidel ,Hello, is MIIM (MDIO) not supported in Nvidia L4T 35+? If I configure fixed-link mode, can the switch KSZ9896 work properly?

lantianhaohh, 2023-01-29:
@sgidel ,Hello, I should ask you like this, my jetson-agx-xavier is L4T 35+, the kernel version is 5.10, ksz9896 is connected to MIIM (MDIO), how to make the switch work normally, the network is normal. Thank you

lantianhaohh, 2023-01-29:
@sgidel ,Hello,Can you guide me to configure Fixed-Link mode in Jetson? I want to try this method first, that is, the supplier said: 

- enable fixed-link, for platforms that support device trees, just add a fixed-link child node to the mac device node, if some platforms do not support device trees, you need to manually register the fixed-link PHY in the mac driver, 
- but also note: fixed-link also needs mac-layer driver support;
- For a switch with a 1000M RGMII interface, you may need to configure RGMII delay on the switch or CPU mac side to ensure the normal operation of RGMII.
But I don't know much about how to configure, can you teach me how to implement the use of ksz? Thank you.

sgidel, 2023-01-30:
Fixed link mode works fine and you can operate the switch in "dumb" switch mode without any drivers at all.
The reason MDIO mode is discouraged is because as far as I know, not all of the internal registers are accessible over MDIO, especially the switch related ones. Its mainly just the phy registers.
For the RGMII delay I physically put a longer trace for TXC and use RGMII-RXID mode. There is an errata on the switch where TXID doesnt work properly. That said, I have tried all of the combinations for phy-mode and they all worked.
Here is what my ethernet node looks like. Note this is for the orin and the only relevant part is the fixed-link section. 
```
eth0: ethernet@2310000 {
		status = "okay";
		nvidia,mac-addr-idx = <0>;
		nvidia,max-platform-mtu = <9000>;
		nvidia,promisc_mode = <1>;
		nvidia,pause_frames = <0>;
		dma-coherent;
		//nvidia,phy-reset-gpio = <&tegra_main_gpio TEGRA234_MAIN_GPIO(G, 5) 0>;
		phy-mode = "rgmii-rxid";
		nvidia,if-name = "ethsw0";
		fixed-link {
			speed = <1000>;
			full-duplex;
		};
	};
```

lantianhaohh, 2023-01-31:
@sgidel ,Hello,After configuring fixed-link mode on your board, does the KSZ9896 work properly? The network can also be pinged, right? This is the path I modified in the Jetson AGX Xavier kernel source: `./hardware/nvidia/platform/t19x/common/kernel-dts/t19x-common-platforms/tegra194-platforms-eqos.dtsi, `
modify this file 'tegra194-platforms-eqos.dtsi' , amend to read as follows:
```
#include <dt-bindings/gpio/tegra194-gpio.h>

/ {
        ethernet@2490000 {
		interrupts =	<0 194 0x4>,    /* common */
				<0 186 0x4>,    /* tx0 */
				<0 190 0x4>;    /* rx0 */
		/* rxq_enable_ctrl = <rx0 rx1 rx2 rx3>
		 * 0x0 = Not enabled, 0x1 = Enabled for AV
		 * 0x2 = Enabled for Legacy, 0x3 = Reserved
		 */
		nvidia,rxq_enable_ctrl = <2>;
		nvidia,num-dma-chans = <1>;
		nvidia,dma-chans = <0>;
		nvidia,num-mtl-queues = <1>;
		nvidia,mtl-queues = <0>;
		nvidia,rx-queue-prio = <0x2>;
		nvidia,tx-queue-prio = <0x0>;
		/* 0=enable, 1=disable */
		nvidia,pause_frames = <0>;
		nvidia,phy-reset-gpio = <&tegra_main_gpio TEGRA194_MAIN_GPIO(G, 5) 0>;
		// phy-mode = "rgmii-id";
		//phy-handle = <&phy>;
		phy-mode = "rgmii-rxid";
		nvidia,max-platform-mtu = <16383>;
		nvidia,if-name = "ethsw0";
		fixed-link {
			speed = <1000>;
			full-duplex;
		};
		// mdio {
		// 	compatible = "nvidia,eqos-mdio";
		// 	#address-cells = <1>;
		// 	#size-cells = <0>;

		// 	phy: phy@0 {
		// 		reg = <0>;
		// 		interrupt-parent = <&tegra_main_gpio>;
		// 		interrupts = <TEGRA194_MAIN_GPIO(G, 4) IRQ_TYPE_LEVEL_LOW>;
		// 		marvell,copper-mode;
		// 		/* Setup LED[2] as interrupt pin (active low) */
		// 		marvell,reg-init = <0x03 0x12 0x7fff 0x880>;
		// 	};
		// };
	};

	thermal-zones {
		CPU-therm {
			trips {
				MAKE_EQOS_TRIP(m40, -40000, 5000);
				MAKE_EQOS_TRIP(m5, -5000, 5000);
				MAKE_EQOS_TRIP(p30, 30000, 5000);
				MAKE_EQOS_TRIP(p65, 65000, 5000);
				MAKE_EQOS_TRIP(p100, 100000, 5000);
			};

			cooling-maps {
				MAP_EQOS(m40, 1);
				MAP_EQOS(m5, 2);
				MAP_EQOS(p30, 3);
				MAP_EQOS(p65, 4);
				MAP_EQOS(p100, 5);
			};
		};
	};
};
```
Is it modified like this? Thank you.

lantianhaohh, 2023-01-31:
@sgidel ,Hello,After I update the device tree, ifconfig to check the network, find that there is no assigned IP and the network is not connected, is there anything else to configure? Can RGMII delay be configured via the device tree?
```
 ~ ifconfig                                                                   
docker0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500                           
        inet 172.17.0.1  netmask 255.255.0.0  broadcast 172.17.255.255          
        ether 02:42:5b:0e:40:3e  txqueuelen 0  (Ethernet)                       
        RX packets 0  bytes 0 (0.0 B)                                           
        RX errors 0  dropped 0  overruns 0  frame 0                             
        TX packets 0  bytes 0 (0.0 B)                                           
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0              
                                                                                
ethsw0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500                    
        ether 48:b0:2d:4d:e0:a6  txqueuelen 1000  (Ethernet)                    
        RX packets 19  bytes 5163 (5.1 KB)                                      
        RX errors 0  dropped 0  overruns 0  frame 0                             
        TX packets 92  bytes 15005 (15.0 KB)                                    
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0   
```

sgidel, 2023-01-31:
yes, uncomment it from the ethernet node as you have it. This should work, but check dmesg for any ethernet related errors. If you used equal length traces keep it as rgmii-id.
Since now the OS thinks a cable is always plugged in, the behavior can be a bit weird and it wont always retry DHCP. You may need to run dhclient after it has booted.

lantianhaohh, 2023-01-31:
@sgidel ,Hello,Is it modified like this?
```
#include <dt-bindings/gpio/tegra194-gpio.h>

/ {
        ethernet@2490000 {
		interrupts =	<0 194 0x4>,    /* common */
				<0 186 0x4>,    /* tx0 */
				<0 190 0x4>;    /* rx0 */
		/* rxq_enable_ctrl = <rx0 rx1 rx2 rx3>
		 * 0x0 = Not enabled, 0x1 = Enabled for AV
		 * 0x2 = Enabled for Legacy, 0x3 = Reserved
		 */
		nvidia,rxq_enable_ctrl = <2>;
		nvidia,num-dma-chans = <1>;
		nvidia,dma-chans = <0>;
		nvidia,num-mtl-queues = <1>;
		nvidia,mtl-queues = <0>;
		nvidia,rx-queue-prio = <0x2>;
		nvidia,tx-queue-prio = <0x0>;
		/* 0=enable, 1=disable */
		nvidia,pause_frames = <0>;
		nvidia,phy-reset-gpio = <&tegra_main_gpio TEGRA194_MAIN_GPIO(G, 5) 0>;
		phy-mode = "rgmii-id";
		//phy-handle = <&phy>;

		nvidia,max-platform-mtu = <16383>;
		nvidia,if-name = "ethsw0";
		fixed-link {
			speed = <1000>;
			full-duplex;
		};
		mdio {
			compatible = "nvidia,eqos-mdio";
			#address-cells = <1>;
			#size-cells = <0>;

			phy: phy@0 {
				reg = <0>;
				interrupt-parent = <&tegra_main_gpio>;
				interrupts = <TEGRA194_MAIN_GPIO(G, 4) IRQ_TYPE_LEVEL_LOW>;
				marvell,copper-mode;
				/* Setup LED[2] as interrupt pin (active low) */
				marvell,reg-init = <0x03 0x12 0x7fff 0x880>;
			};
		};
	};

	thermal-zones {
		CPU-therm {
			trips {
				MAKE_EQOS_TRIP(m40, -40000, 5000);
				MAKE_EQOS_TRIP(m5, -5000, 5000);
				MAKE_EQOS_TRIP(p30, 30000, 5000);
				MAKE_EQOS_TRIP(p65, 65000, 5000);
				MAKE_EQOS_TRIP(p100, 100000, 5000);
			};

			cooling-maps {
				MAP_EQOS(m40, 1);
				MAP_EQOS(m5, 2);
				MAP_EQOS(p30, 3);
				MAP_EQOS(p65, 4);
				MAP_EQOS(p100, 5);
			};
		};
	};
};
```

sgidel, 2023-01-31:
Remove the mdio block.
Should look like this:
```
#include <dt-bindings/gpio/tegra194-gpio.h>

/ {
        ethernet@2490000 {
		interrupts =	<0 194 0x4>,    /* common */
				<0 186 0x4>,    /* tx0 */
				<0 190 0x4>;    /* rx0 */
		/* rxq_enable_ctrl = <rx0 rx1 rx2 rx3>
		 * 0x0 = Not enabled, 0x1 = Enabled for AV
		 * 0x2 = Enabled for Legacy, 0x3 = Reserved
		 */
		nvidia,rxq_enable_ctrl = <2>;
		nvidia,num-dma-chans = <1>;
		nvidia,dma-chans = <0>;
		nvidia,num-mtl-queues = <1>;
		nvidia,mtl-queues = <0>;
		nvidia,rx-queue-prio = <0x2>;
		nvidia,tx-queue-prio = <0x0>;
		/* 0=enable, 1=disable */
		nvidia,pause_frames = <0>;
		/* no reset GPIO */
		//nvidia,phy-reset-gpio = <&tegra_main_gpio TEGRA194_MAIN_GPIO(G, 5) 0>;
		phy-mode = "rgmii-id";
		nvidia,max-platform-mtu = <16383>;
		nvidia,if-name = "ethsw0";
		fixed-link {
			speed = <1000>;
			full-duplex;
		};
	};

	thermal-zones {
		CPU-therm {
			trips {
				MAKE_EQOS_TRIP(m40, -40000, 5000);
				MAKE_EQOS_TRIP(m5, -5000, 5000);
				MAKE_EQOS_TRIP(p30, 30000, 5000);
				MAKE_EQOS_TRIP(p65, 65000, 5000);
				MAKE_EQOS_TRIP(p100, 100000, 5000);
			};

			cooling-maps {
				MAP_EQOS(m40, 1);
				MAP_EQOS(m5, 2);
				MAP_EQOS(p30, 3);
				MAP_EQOS(p65, 4);
				MAP_EQOS(p100, 5);
			};
		};
	};
};
```

lantianhaohh, 2023-01-31:
@sgidel ,Hello,After I update the device tree, ifconfig to check the network, find that there is no assigned IP and the network is not connected, 
```
ifconfig
docker0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 172.17.0.1  netmask 255.255.0.0  broadcast 172.17.255.255
        ether 02:42:9d:ea:a2:78  txqueuelen 0  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

dummy0: flags=195<UP,BROADCAST,RUNNING,NOARP>  mtu 1500
        inet6 fe80::6864:5ff:fe46:1feb  prefixlen 64  scopeid 0x20<link>
        ether 6a:64:05:46:1f:eb  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 39  bytes 11706 (11.7 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

ethsw0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ether 48:b0:2d:4d:e0:a6  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 115  bytes 24213 (24.2 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

l4tbr0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 4e:fa:f3:20:18:9d  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 307  bytes 24858 (24.8 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 307  bytes 24858 (24.8 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

rndis0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 4e:fa:f3:20:18:9d  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 4e:fa:f3:20:18:9f  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.187  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::7866:ac04:3dc:9002  prefixlen 64  scopeid 0x20<link>
        ether 28:c6:3f:19:25:d5  txqueuelen 1000  (Ethernet)
        RX packets 897  bytes 246703 (246.7 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 349  bytes 73837 (73.8 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

lantianhaohh, 2023-01-31:
```
➜  ~ sudo dhclient
[sudo] password for nvidia: 
RTNETLINK answers: File exists
```

lantianhaohh, 2023-01-31:
@sgidel ,Hello,check dmesg for any ethernet related errors:
```
sudo dmesg|grep ethernet
[    9.025100] nvethernet 2490000.ethernet: Adding to iommu group 30
[    9.033193] nvethernet 2490000.ethernet: failed to read skip mac reset flag, default 0
[    9.047118] nvethernet 2490000.ethernet: failed to read MDIO address
[    9.060828] nvethernet 2490000.ethernet: setting to default DMA bit mask
[    9.073241] nvethernet 2490000.ethernet: set default TXQ to TC mapping
[    9.073248] nvethernet 2490000.ethernet: Setting default PTP RX queue
[    9.088454] nvethernet 2490000.ethernet: failed to get phy reset gpio error: -2
[    9.100064] nvethernet 2490000.ethernet: failed to get eqos_rx_m clk
[    9.120321] nvethernet 2490000.ethernet: failed to get eqos_rx_input clk
[    9.120327] nvethernet 2490000.ethernet: failed to get eqos_tx_divider clk
[    9.143235] nvethernet 2490000.ethernet: Ethernet MAC address: 48:b0:2d:4d:e0:a6
[    9.173077] nvethernet 2490000.ethernet: Macsec not supported
[    9.187941] nvethernet 2490000.ethernet: ethsw0 (HW ver: 50) created with 1 DMA channels
[   12.450912] using random self ethernet address
[   12.451951] using random host ethernet address
[   13.324659] nvethernet 2490000.ethernet ethsw0: Link is Up - 1Gbps/Full - flow control off
[   13.439665] using random self ethernet address
[   13.446428] using random host ethernet address
```

sgidel, 2023-01-31:
Many of those can be ignored but the eqos_xx_clk errors are concerning. That sounds like something is wrong with the device tree other than the fixed link setup. You may get better support for that issue on Nvidia's forum and this is starting to get out of scope of this repo.

lantianhaohh, 2023-01-31:
@sgidel ,Okay, I'll also go to the NVIDIA official forum and ask, thank you

lantianhaohh, 2023-01-31:
@sgidel ,Is this eqos_xx_clk error caused by not setting RGMII delay?

lantianhaohh, 2023-01-31:
@sgidel ,Hello,This is a topic created at NVIDIA, can you help take a look? If something is wrong, you can correct my mistake, thank you，[https://forums.developer.nvidia.com/t/use-the-ksz9896-phy-to-replace-the-jetson-agx-xavier-phy-driver-via-the-mdio-interface/241103/18](https://forums.developer.nvidia.com/t/use-the-ksz9896-phy-to-replace-the-jetson-agx-xavier-phy-driver-via-the-mdio-interface/241103/18)