Microchip-Ethernet/EVB-KSZ9477 #93: KSZ9477 can not enter power saving mode

rolandash, 2023-05-31:
Dear Microchip,

On my customization board I am trying to set 9477 entering one of the power saving mode, either EDPD or soft power down mode, from an external MCU. The control interface is I2C, and all enet ports are not used when doing the tests.

Following the instructions in 9477 datasheet, I write '01' or '10' to bits [4:3] into the Power Down Control 0 Register. 

The write operation seems to be succeeded, as I can read the expected bits back from the register. However, the chip does not seem actually get into the power saving mode, as I can see the current measurement from the meter does not change. The result is the same for either EDPD or soft power down mode.

So do I need to do anything else other than just writing the register? or any special setup in hardware I have missed?

Thank you.

/Rolandr soft power down mode.

So do I need to do anything else other than just writing the register? or any special setup in hardware I have missed?

Thank you.

/Roland

rolandash, 2023-06-02:
Some further test results:

1. I followed module 9 in errata document setting the MMD registers, and get some difference. The chip initially consumes about 260mA@5V at power up. After module 9 is applied, the current consumption drops little bit to ~230mA.
2. The I followed up with register writings to set the chip into EDPD (power mode 1), and this time I get another difference, as the current drops further down to ~200mA. It is still quite high but at least got some progress.
3. The attempt to set the chip into soft power down mode (mode 2) still not working, no matter module 9 is applied or not.

I still do not have the full picture behind it yet, but seems power saving mode has something to do with those MMD registers. Unfortunately I can not find any further detailed instructions in documents I can get publicly, so I do not know how to  proceed.

Can you please help with it?

Thank you.

/Roland

Ravi-Hegde, 2023-06-02:
Hi Ronald, Could you log a ticket @ https://microchip.my.site.com/s/ ? Looks like need to get HW support team help.

Ravi-Hegde, 2023-06-02:
Sorry I was not clear. Here please: https://microchip.my.site.com/s/supportservice "Create Microchip Support Case".

rolandash, 2023-06-06:
@Ravi-Hegde thank you. I have filed a case through the portal. 
It seems that it may took a while before an engineer can be assigned to answer this case. I would like to take this period of time trying to explore the related portion of linux driver for EVB9477. 
I can see that after the SoC has configured 9477 through SPI interface, the current 9477 consumed significantly drops down. However, when I check into the source code by searching register names/macros, it seems not obvious where the power management related code can be found. 
Can you please guide me where the 9477 power setting related code is located?

rolandash, 2023-06-10:
I loaded the complete KSZ9477 linux driver project source code.

The global power control register 0 is defined in KSZ9477_reg.h,

````
#define REG_SW_POWER_MANAGEMENT_CTRL	0x0201
#define SW_POWER_DOWN_MODE		0x3
#define SW_ENERGY_DETECTION		1
#define SW_SOFT_POWER_DOWN		2
#define SW_POWER_SAVING			3
````

But these macros never been referenced in source code, neither the EDPD power mode or soft power down mode.  Only 8795 driver has referenced similar macros. 

So it looks like 9477/9897 driver never accessed global power control register 0 (0x0201). In other words, the global power control function of 9477/9897 is not verified by official software driver on 9477 EVB. 

Can you please confirm it?

Thank you.

/Roland

triha2work, 2023-06-15:
The definitions in ksz9897.h are incorrect.  However, writing 0x10 does accomplish something like stopping IBA operation and dropping link on all ports.  Reading any register will immediately wake the system up and the links come back.

rolandash, 2023-06-23:
@triha2work 
Thank you for the reply. 
Not sure I understood you clearly.
1. if definitions in ksz9897.h are not correct, then what are the correct ones? Are you implying some other registers should be used?
2. behavior of writing 0x01 your described seems still different from datasheet, as quoted from page 85. in below:
````
After powering down the entire chip, exit power down by first setting this register to normal operation, and then set the SGMII register to normal operation.
````
By my understanding I need to write 0x00 back to this specific register to bring the chip up again, rather than any register?

triha2work, 2023-06-24:
The definition of shifting the power down bits is not there, so programing the register using that information is wrong.  But because that register is never used in the driver so it can be said the definition is incomplete.
I am not sure the datasheet description of power down is correct, as I observed different behavior.  As mentioned before writing 0x10 seems to power down the chip, as the IBA operation immediately stops and the 0x104 register is changed to a random value.  It does not seem all registers are reset.  Because the driver switches back to SPI access and any read wakes up the chip, reading the register 0x201 never gets the value 0x10.  Writing the correct value to 0x104 gets the IBA operation running again.
In U-Boot writing 0x10 makes all ports drop link.  Reading the register again makes them come up again.

rolandash, 2023-06-26:
@triha2work 
thank you for the detailed information.

I think you are correct about the behavior of writing 0x10 into 0x201, as I can also reproduce it on my own hardware, after removed the register read-back line of code. The current consumption now drops to about 100mA, and stand there in stable, as if some kind of "current floor" there. 

I do not know what  is really consuming this 100mA now. From the power aspect this is not really a shut down but more like a reset. 

Anyway, I was planning to use this soft-power-down feature to save one GPIO of MCU as board power management. Now it looks like not a good idea as it might get into an uncertain state.