Microchip-Ethernet/EVB-KSZ9477 #95: Jetson agx orin  RGMII spi-ksz9893 :ubuntu login: [   28.502955] [eqos_poll_for_swr][598][type:0x4][loga-0x0] poll_for_swr: timeout [   28.510452] net ethsw0: failed to poll MAC Software reset

Jianxiong-Chen, 2023-06-16:
chosen {
		nvidia,ether-mac0 = "48:B0:2D:5D:16:18";
		nvidia,ether-mac = "48:B0:2D:5D:16:18";
	};

	eth0: ethernet@2310000 {
		status = "okay"; 
		nvidia,pause_frames = <0>;
		//phy-mode = "rgmii-id";
		//nvidia,max-platform-mtu = <16383>;
		//no reset GPIO 
		#nvidia,phy-reset-gpio = <&tegra_main_gpio TEGRA234_MAIN_GPIO(G, 5) 0>;

		nvidia,mac-addr-idx = <0>;
		nvidia,max-platform-mtu = <9000>;
		nvidia,promisc_mode = <1>;
		dma-coherent;
		phy-mode = "rgmii-rxid";
		nvidia,if-name = "ethsw0";

		fixed-link {
			speed = <1000>;
			full-duplex;
		};

	
	};
	

	spi1: spi@c260000 {
		status = "okay";
		spi-max-frequency = <25000000>;
		ksz9893: ksz9893@0 {
		compatible = "microchip,ksz9893";
		#pinctrl-names = "default";
		reg = <0>;
		phy-mode = "rgmii-id";
		status = "okay";
		spi-max-frequency = <10000000>;
		//spi-cpha;
		//spi-cpol;	
		//fsl,rgmii_txc_dly;
		//interrupt-parent = <&tegra_main_gpio>;
		//interrupts = <TEGRA234_MAIN_GPIO(G, 4) IRQ_TYPE_LEVEL_LOW>;
		//nvidia,phy-reset-gpio = <&tegra_main_gpio TEGRA234_MAIN_GPIO(G, 5) 0>;
			ports {
				#address-cells = <1>;
				#size-cells = <0>;
				port@0 {
					reg = <0>;
					label = "lan1";					
				};
				port@1 {
					reg = <1>;
					label = "lan2";
				};
				port@3 {
					reg = <2>;
					label = "cpu";
					ethernet = <&eth0>;
					fixed-link {
						speed = <1000>;
						full-duplex;
					};
				};
			};
		};	
	};

ubuntu login: [   28.502955] [eqos_poll_for_swr][598][type:0x4][loga-0x0] poll_for_swr: timeout
[   28.510452] net ethsw0: failed to poll MAC Software reset

Jianxiong-Chen, 2023-06-16:
[image]
[image]

Jianxiong-Chen, 2023-06-16:
[image]

triha2work, 2023-08-01:
As the DSA driver is being used and there is no required modification to the MAC driver I am not sure how the switch driver can affect the MAC driver.  So the MAC driver can run fine by itself without loading the switch driver?
There is probably an error in the device tree where port@3 should be port@2, but it may not matter.