Microchip-Ethernet/EVB-KSZ9477 #99: KSZ8563 - get strange results, how to use properly?

starbrights, 2023-08-02:
Using the KSZ8653 (SPI) and try to figure out how it works (5.14 kernel). 
With no readme I start digging into source code, so I found that I have to call some path with "registers" in and found that 3 folders that seems to give access to registers:
```
./kernel/debug/regmap/spi1.0-32/registers
./kernel/debug/regmap/spi1.0-16/registers
./kernel/debug/regmap/spi1.0-8/registers
```
Which one to call - some reg are 8, some are 16bit others even 32bit?

regs_bin /sys/kernel/debug/regmap/spi1.0-8 ?

But whatever I handover (-8, -16 or -32) as param it uses:
reg_size: 4
reg_cnt: 8
reg_fmt %08x
and a simple r, rb, rw, rd (which I think should access 8, 16 or 32bit) doesn't work.
Only when I give a count (?)

I get only output with:
```
reg> r 1 3
0001: 30303030 203a3130
```
which is completly nonsense.

When I "cat" the /sys/kernel/debug/regmap/spi1.0-8/registers I get an uncomfortable list of all registers, so at least read access should work. 
```
cat /sys/kernel/debug/regmap/spi1.0-8/registers
000000: 00
000001: 98
000002: 93
```
But finaly I need the write access to MMD register 0x3c to set/clear EEE. Will that work with this tool (once I get a working read)? Or do you know an other way to do so?

triha2work, 2023-08-03:
I am not sure those created "registers" files are for the standard register access API.  The regs_bin utility were created for that implementation inside the non-DSA switch driver.  That API allows the registers file to be opened for read and write access.  The file location indicates the register and the count indicates how many to read or write and there is a buffer to hold the result.  I do not think using the "cat" command works on that properly.  Anyway I will try those files to see what is going on.  The KSZ switch access is always 8-bit.  There are 16-bit and 32-bit accesses because some registers are defined in those widths.

triha2work, 2023-08-03:
I just tried using "cat" on the "registers" file created by non-DSA switch driver.  It just displayed non-ASCII characters.  That API deals with binary numbers so that is expected.  The "registers" file created by regmap displays nice ASCII characters and so it uses a different system.  It seems to behave like the old debugfs used by the non-DSA driver of old KSZ switches.  Those switches have a limited set of registers like 256, so it is okay to dump those registers.  But KSZ8563 is a new switch with different register definitions are there are a lot of holes in those registers.  Dumping all is not a good idea.
If you really want to access those switch registers there is a patch to easily add the registers API to the DSA driver so that regs_bin can work with it.  But you need to specify the exact version (5.14) as those regmap implementations keep changing.

starbrights, 2023-08-03:
Do I get it right, this tool is for non-DSA drivers only? But as far as I know only the DSA driver is mainline - at least for that chip.
My origin problem is that I need to disable EEE as it auses trouble.
[https://microchip.my.site.com/s/article/Link-instability-issue-in-KSZ8563-when-EEE-feature-enabled](url)

```
ethtool --show-eee lan1
netlink error: Operation not supported

```
And I can't disable AutoNeg (as I need it) to disable EEE too. 
This register for EEE is an indirect Register, only accessible via multiple writes to MMD Setup/Data Registers.