Microchip-Ethernet/EVB-KSZ9477 #104: Build issue with Ubuntu 22.04.3 LTS

arthurbenemann, 2024-02-08:
Tring to build current master branch bf22b0ed09e537d6b760d3f38a0eff59513979d9, on a fresh Ubuntu server and following the build instructions for SD card build, fails with the following message:

``` Bash
arthur@ptp1:~/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.3 LTS
Release:        22.04
Codename:       jammy
arthur@ptp1:~/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot$ uname -a
Linux ptp1 5.15.0-92-generic #102-Ubuntu SMP Wed Jan 10 09:33:48 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
arthur@ptp1:~/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot$ git log | head -n 1
commit bf22b0ed09e537d6b760d3f38a0eff59513979d9
arthur@ptp1:~/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot$ make
>>> host-m4 1.4.18 Building
PATH="/home/arthur/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/host/bin:/home/arthur/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/host/sbin:/home/arthur/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/host/usr/bin:/home/arthur/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/host/usr/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" PKG_CONFIG="/home/arthur/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/host/usr/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_LIBDIR="/home/arthur/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/host/usr/lib/pkgconfig"  /usr/bin/make -j3  -C /home/arthur/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/build/host-m4-1.4.18/
/usr/bin/make  all-recursive
Making all in .
make[4]: Nothing to be done for 'all-am'.
Making all in examples
make[4]: Nothing to be done for 'all'.
Making all in lib
/usr/bin/make  all-am
  CC       c-stack.o
  CC       cloexec.o
  CC       close-stream.o
  CC       closein.o
In file included from /usr/include/signal.h:328,
                 from ./signal.h:52,
                 from c-stack.c:49:
c-stack.c:55:26: error: missing binary operator before token "("
   55 | #elif HAVE_LIBSIGSEGV && SIGSTKSZ < 16384
      |                          ^~~~~~~~
make[5]: *** [Makefile:1915: c-stack.o] Error 1
make[5]: *** Waiting for unfinished jobs....
make[4]: *** [Makefile:1674: all] Error 2
make[3]: *** [Makefile:1572: all-recursive] Error 1
make[2]: *** [Makefile:1528: all] Error 2
make[1]: *** [package/pkg-generic.mk:204: /home/arthur/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/build/host-m4-1.4.18/.stamp_built] Error 2
make: *** [Makefile:41: _all] Error 2

```

embeddman, 2024-02-09:
This buildroot version should be built under Ubuntu 16.04

arthurbenemann, 2024-02-09:
Thank you @embeddman !

With 16.04, it failed after a while resulting in:
```
...
checking for uuid_generate in -luuid... yes
Disabling private uuid library
checking pkg-config is at least version 0.9.0... yes
checking for blkid_get_cache in -lblkid... no
configure: error: external blkid library not found
package/pkg-generic.mk:191: recipe for target '/home/arthur/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/build/host-e2fsprogs-1.43.9/.stamp_configured' failed
make[1]: *** [/home/arthur/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/build/host-e2fsprogs-1.43.9/.stamp_configured] Error 1
Makefile:41: recipe for target '_all' failed
make: *** [_all] Error 2
```

Got it to build by applying the instructions at https://www.linux.org/threads/configure-error-external-blkid-library-not-found.30906/#post-103317

closed 2024-02-09 by arthurbenemann