Microchip-Ethernet/EVB-KSZ9477 #105: Issues programming NAND on EV-KSZ9477

arthurbenemann, 2024-02-17:
Following the EVB_KSZ9477_Source_Build_Instructions for programming the SDcard worked, but I'm not able to programm the on-board NAND using those instructions.

When running the NANDprogramming instructions (step9),  it locks up at the `lowlevelinit` step of programming, as shown in the snippet below:

```
arthur@testbuild:~/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot$ sudo ./flash_board_x64
Executing: sama5d3_xplained O=/home/arthur/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/images /home/arthur/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/host/opt/sam-ba/sam-ba /dev/ttyUSB0 at91sama5d3x-xplained ./board/atmel/nandflash.tcl -- sama5d3_xplained at91-sama5d3_xplained.dtb video=LVDS-1:800x480-16
-I- Waiting ...
-I- TCL platform : Linux
-I- SAM-BA CDC 2.16  on : linux
-I- Retrieved arguments from command line :
-I- argv 0 : /dev/ttyUSB0
-I- argv 1 : at91sama5d3x-xplained
-I- argv 2 : ./board/atmel/nandflash.tcl
-I- Connection /dev/ttyUSB0
-I- Connection : /dev/ttyUSB0 (target(comType) = 0)
-I- Board : at91sama5d3x-xplained
-I- Traces Level : 4
-I- target(handle) : file5
-I- sourcing board description file /home/arthur/EVB-KSZ9477/KSZ/Atmel_SOC_SAMA5D3/buildroot/output/host/opt/sam-ba/tcl_lib/at91sama5d3x-ek/at91sama5d3x-ek.tcl
Read device Chip ID at 0xFFFFEE40 --- get 0xffffffff8a5c07c3
-I- Found chip : at91sama5d3x (Chip ID : 0xffffffff8a5c07c3)
-I- Loading applet applet-lowlevelinit-sama5d3x.bin at address 0x308000

```

This happens with either a serial cable connected to J10 (`ttyUSB0`), or a USB cable to J12 (`ttyACM0`). Even after 1 hour it is still stuck on the same step.

Thanks in advance for any help.

triha2work, 2024-02-20:
Is this a new problem?  Or did it work before?

triha2work, 2024-02-22:
It seems the new KSZ9477 evaluation board has a hardware change that makes the old sam-ba utility not able to program the NAND.  Even if you can avoid the stuck issue by restarting the NAND programing procedure the script will stop during processing the nandflash.tcl script.  You will need to use the new sam-ba utility provided by the newer Buildroot hosted in linux4sam repository.  Unfortunately that new sam-ba utility uses a different script and procedure and so requires additional step to generate the U-Boot environmental variable block.  If you know how to do that manually with the mkenvimage utility then it may be simple to use the new sam-ba parameters to invoke the NAND programming procedure.
I am in the process of creating a new build procedure using a new Buildroot like buildroot-at91-linux4sam-2022.10 because the current Buildroot environment no longer works in new Linux distros like 2023.10.  If you can wait for that then you can use that to build Linux 4.14, 4.19, 5.4, 5.10, 5.15, and 6.1 kernels.