Microchip-Ethernet/EVB-KSZ9477 #109: AVB Packets dropped during extended, high traffic use

alexbross, 2024-12-05:
Hello,

I'm currently working through an issue where sometime in a two hour long capture of high load AVB traffic we get packet drops. We do not see this with the MOTU switch. 

I'm looking to better understand the implications of using two step PTP versus one step. As I understand from:

https://github.com/Microchip-Ethernet/EVB-KSZ9477/issues/83

Two step PTP looks like a no go. However, the tools we're using this might not support one step modes. 

Could the two step clock be the issue? As far as I can tell if a node restarts it doesn't effect the rest of the network.

Thanks!

alexbross, 2024-12-05:
Additionally I was hoping for clarification on module 20 from here:

https://ww1.microchip.com/downloads/en/DeviceDoc/80000754C.pdf

Is implemented in the code base or is this something that needs to be implemented?

triha2work, 2025-01-09:
The KSZ9477 switch is no longer recommended for use with AVB as the 2-step operation has a bug that causes PTP messages to be dropped, and that is not acceptable.  AVB traffic probably has dropped packets, but it was not verified.  There is a workaround to use 1-step clock to simulate 2-step operation, but it is probably not acceptable for gPTP if the PTP messages need to adhere to the standard.

alexbross, 2025-01-09:
Thanks for the reply. I am on your thoughts about the ramifications of some of the work arounds floating around. It seems like we should move to a new IC all together or switch our system to one step mode. This brings up another question, would Microchip still recommend the IC if the system was in one step mode? 

Specifically, the MOTU AVB switch uses the KSZ956R and has the same errata module.

https://ww1.microchip.com/downloads/en/DeviceDoc/80000755C.pdf

That said, after testing their switch with our system, their work around seems quite solid. Digging into how they fixed it, their fix is to use it in one step mode and emulate the follow up messages.

> To avoid this issue we do not actually enable two-step mode in the switch hardware (we leave it in one-step mode). The only consequence of doing this is that correction field in the PDelay_Resp message is filled in by the switch hardware as it should be in one step mode. We then manually send a PDelay_Resp_Follow_Up message with the correction field carefully calculated to cancel out the correction field from the initial response.

Our workaround avoids this issue my making sure that all the fields in the PDelay_Resp message and PDelay_Resp_Follow_Up message are identical to what would be expected from properly functioning two-step hardware. The only difference is that now the correction fields are non-zero in both messages. What this means, is that for peers that just ignore the correction fields altogether, they will continue to be able to compute the correct peer delay value. Additionally, for peers that incorporate the correction fields into their computation (using the equation from 802.1AS), they will also compute the correct value, because the correction field in the follow-up message is filled in with a value that will cancel out the correction field in the initial response.

alexbross, 2025-12-03:
> The KSZ9477 switch is no longer recommended for use with AVB as the 2-step operation has a bug that causes PTP messages to be dropped, and that is not acceptable. AVB traffic probably has dropped [...]

Hello, figured this is still the best place to continue this conversation.

The errata seems to have an update, specifically:

"To work around this issue, a Software Two-Step Simulation Mode in hardware 1-Step Mode is suggested. For gPTP operation, the switch hardware needs to be configured in 1-Step Mode by setting bit 0 of Global PTP Message Config 1 Register (Bit 0 of 0x0514 - 0x0515). Because the switch hardware is running in 1-Step Mode, the hardware is still automatically updating the contents of the Sync and Pdelay_Resp messages. Sync, Follow-up: The Sync transmit interrupt needs to be enabled so that Follow_Up messages can be updated with the Sync transmit timestamp. The master sends a Two-Step Sync, which contains the Sync transmit timestamp. The master then sends a Follow_Up with the same transmit timestamp. The slave still receives the Sync transmit timestamp from the Follow_Up, so there is no change in application."

To do this for the 1-step mode, I go:

regs_bin spi1.0
>wb 0x515 fd
>rb 0x515 which prints (fd)

This does seem to do part of the job.

However, how do I enable transmit sync interrupt? Do I have to do it for every AVB port? I'm a bit confused what register it's referring to.

Please let me know if there's a better way to do this, but ideally, I do a trial run with a clean build and do this post boot.

Thanks!