Microchip-Ethernet/EVB-KSZ9477 #115: KSZ8463 Linux 6.6 PTP path delay negative

paulfuchs, 2025-07-03:
I am facing a problem regarding PTP and a negative path delay with the KSZ8463 under Linux version 6.6.
Target System: Zynq-7000 with Macb and KSZ8463 drivers integrated.

The Zynq is the grandmaster.
I run the following commands to start linuxptp:
```
./ptp/linuxptp/main/ptp4l -i eth0 -2 -f ptp/linuxptp/main/default.cfg
```
and also
```
./ptp/linuxptp/main/phc2sys -r -d /dev/pps0 -s /dev/ptp0 -w
```

The Master correctly outputs sync and announce messages. Furthermore it also reacts
to delay request messages with a delay reponse.
The seconds of the timestamps in the delay response however are wrong after a certain time, compared to those of the sync message.
After a few minutes of running the drift is already -30 seconds and becoming more (see wireshark screenshots).
That makes no sense to me, because the delay request was sent after the Sync message and therefore the "receiveTimestamp" in the delay reponse should be greater (or the same seconds) as in the sync message.

[image]

[image]

So either my local clock does not synchronize to the phc of the switch or there is another issue.
What could be the reason for that behaviour?

triha2work, 2025-07-03:
KSZ8463 is the master clock but does the ordinary clock connect directly to KSZ8463 switch?  Why there is a correction field in the Sync message as if it goes through a 1-step TC?
There may be a bug in software.  The PTP clock is reflected in the Sync transmit timestamp, but the receive timestamp only contains 2 bits of second value, so the driver needs to update the full second with the one it maintains.  Typically this second is updated every time the PPS signal is detected.  If not the driver reads the PTP clock every second.  Generally when this problem happens the time is off by 4 seconds.  The PTP stack should immediately notifies something is amiss and in verbose mode you should see more errors.  You should run the PTP stack with "-m" parameter.  The phy2sys operation in this case does not matter.

paulfuchs, 2025-07-21:
Hi, yes that's the problem. The receive timestamp is smaller than in the following sync timestamp, what causes a negative path delay. And after half an hour it's already -120 seconds and getting worse.
Do you got any idea where that problem could come from?

triha2work, 2025-07-22:
When this happens does not the PTP stack declare something is faulty and restart the synchronization?  How is the clock allowed to be drifted for so long?  Can you upload a log of the PTP stack in verbose mode when this problem happens?
The only explanation I have is somehow the receive timestamp may not come from KSZ8463 but from the internal MAC which has PTP function.
If you have access to the receiving MAC it is easy to dump the received PTP message and display its header where its 32-bit reserved field contains the receive timestamp.  This timestamp has 30-bit nanosecond and 2-bit second.  You can easily verify the nanosecond part in the Delay_Req to see if it matches the reported timestamp in Delay_Resp.

closed 2025-11-20 by paulfuchs