Microchip-Ethernet/EVB-KSZ9477 #116: KSZ9477 The IGMP function cannot take effect.

bao1007450025, 2025-07-06:
Set reg 0x0370  to 1，However, the IGMP port can still receive forwarded multicast packets.

Is any additional configuration required?

triha2work, 2025-07-07:
IGMP snooping is bit 6.  The behavior is IGMP packets are only forwarded to the host port, which kind of defeat the purpose of using IGMP.

bao1007450025, 2025-07-07:
How to enable multicast filtering and discard unknown packets

bao1007450025, 2025-07-09:
Set to 1, and enable IGMP snooping function. Multicast packets cannot be forwarded normally.

triha2work, 2025-07-22:
The switch has a reserved multicast address table which forwards certain multicast addresses in certain ways.  It has a forward unknown multicast function which can block all unknown multicast addresses.  To filter multicast address the dynamic MAC table has to be programmed to forward that multicast address to certain ports.
Note that for the host its MAC has its own multicast filtering.  Typically its all multicast function is not enabled.  However, it seems the kernel network stack does not send multicast filtering command to the MAC to allow incoming IGMP messages so the kernel IGMP snooping function is not working.  Probably it relies on setting the MAC to promiscuous mode.
The non-DSA switch driver hosted here tries to allow certain multicast addresses while setting the MAC to either all multicast or promiscuous mode and uses software to filter multicast address for different network devices.  This function is only enabled when AVB mode is set.  It is okay to remove this restriction as AVB mode is no longer used by default as there are issues running some operations.

bao1007450025, 2025-07-23:
I am using is non-DSA switch driver.

The following is the activation of IGMP_SNOOP,But PORT2 can still receive multicast messages.

bao1007450025, 2025-07-23:
PORT1 PLC  DEVICE(SEND IGMP)
PORT2 PC
PORT5 device send multicast messages.

PORT2  log
Ethernet II, Src: SiboTechAuto_1f:1e:46 (64:ea:c5:1f:1e:46), Dst: IPv4mcast_40:03:c0 (01:00:5e:40:03:c0)
Internet Protocol Version 4, Src: 192.168.1.23, Dst: 239.192.3.192
User Datagram Protocol, Src Port: 2222, Dst Port: 2222
Common Industrial Protocol, I/O
Common Industrial Protocol, I/O

bao1007450025, 2025-07-24:
Is multicast IP filtering on the host implemented through the ALU, or does it need to be done using ACL?