Microchip-Ethernet/EVB-KSZ9477 #118: [EVB-KSZ9477] Request: Update KSZ9477 driver and i.MX FEC support for Linux kernel 6.12

erhnam, 2025-08-19:
Hello,

I am working with the EVB-KSZ9477 switch and the NXP i.MX platform (FEC Ethernet controller).
After upgrading to Linux kernel 6.12, I have found that the current KSZ9477 driver and the Freescale/NXP FEC integration require updates to properly support this kernel version.

Observed issues:

Kernel crashes when invoking fec_enet_update_ethtool_stats() during initialization.

Problems detecting and initializing multiple Ethernet interfaces (e.g., ethX).

Driver dependencies on older kernel APIs that have changed in 6.12.

Request:
Please update the KSZ9477 driver and the i.MX FEC integration to ensure compatibility with Linux 6.12.

Thank you for your support in keeping KSZ9477 + i.MX platforms up to date with the latest kernel releases.

triha2work, 2025-08-19:
Please look in the EVB-LAN9646 repo also hosted here.  There is a patches directory in ung_apps_external.  The linux-linux4microchip-2025.04 directory holds the patches for the modified 6.12 kernel.  The 006-freescale.patch works on generic 6.12 kernel.

erhnam, 2025-08-20:
Thank you, I will try it!.